llvm/circt #3127: affine-to-staticlogic -> Assertion `isSingleConstant() && "map must have a single constant result

A-Dane, 2022-05-13:
````
// This is a code snippet
for (int i = 1; i <= n; i++)
````

Result in this affine snippet
````
// This is a code snippet creates issues for affine-to-staticlogic - see #map
%1 = affine.for %arg1 = 1 to **#map()[%0]** iter_args(%arg2 = %c0_i32) -> (i32) {
````

This command
````
circt-opt --allow-unregistered-dialect --convert-affine-to-staticlogic -o=staticlogic.txt affine.txt
````

Creates:

circt-opt: /mnt/scratch/tools/github/circt/llvm/mlir/lib/IR/AffineMap.cpp:282: int64_t mlir::AffineMap::getSingleConstantResult() const: Assertion `isSingleConstant() && "map must have a single constant result"' failed.
PLEASE submit a bug report to https://github.com/llvm/llvm-project/issues/ and include the crash backtrace.
Stack dump:
0.      Program arguments: /mnt/scratch/tools/github/circt/build/bin/circt-opt --allow-unregistered-dialect --convert-affine-to-staticlogic -o=staticlogic.txt affine.txt
 #0 0x00007fdcef1cddd5 PrintStackTraceSignalHandler(void*) Signals.cpp:0:0
 #1 0x00007fdcef1cb454 SignalHandler(int) Signals.cpp:0:0
 #2 0x00007fdceea3a520 (/lib/x86_64-linux-gnu/libc.so.6+0x42520)
 #3 0x00007fdceea8ea7c __pthread_kill_implementation ./nptl/./nptl/pthread_kill.c:44:76
 #4 0x00007fdceea8ea7c __pthread_kill_internal ./nptl/./nptl/pthread_kill.c:78:10
 #5 0x00007fdceea8ea7c pthread_kill ./nptl/./nptl/pthread_kill.c:89:10
 #6 0x00007fdceea3a476 gsignal ./signal/../sysdeps/posix/raise.c:27:6
 #7 0x00007fdceea207f3 abort ./stdlib/./stdlib/abort.c:81:7
 #8 0x00007fdceea2071b _nl_load_domain ./intl/./intl/loadmsgcat.c:1177:9
 #9 0x00007fdceea31e96 (/lib/x86_64-linux-gnu/libc.so.6+0x39e96)
#10 0x00007fdcef2f3639 mlir::AffineMap::getSingleConstantResult() const (/mnt/scratch/tools/github/circt/llvm/build/./lib/libMLIRIR.so.15git+0xd0639)
#11 0x00007fdcefd6ff98 mlir::AffineForOp::getConstantUpperBound() (/mnt/scratch/tools/github/circt/llvm/build/./lib/libMLIRAffine.so.15git+0x52f98)
#12 0x00005589e88ab776 (anonymous namespace)::AffineToStaticLogic::createStaticLogicPipeline(llvm::SmallVectorImpl<mlir::AffineForOp>&) AffineToStaticLogic.cpp:0:0
#13 0x00005589e88b10da (anonymous namespace)::AffineToStaticLogic::runOnOperation() AffineToStaticLogic.cpp:0:0
#14 0x00007fdcef45a41a mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) (/mnt/scratch/tools/github/circt/llvm/build/./lib/libMLIRPass.so.15git+0x2c41a)
#15 0x00007fdcef45aab0 mlir::detail::OpToOpPassAdaptor::runPipeline(llvm::iterator_range<llvm::pointee_iterator<std::unique_ptr<mlir::Pass, std::default_delete<mlir::Pass> >*, mlir::Pass> >, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) (/mnt/scratch/tools/github/circt/llvm/build/./lib/libMLIRPass.so.15git+0x2cab0)
#16 0x00007fdcef4593cb mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool) (/mnt/scratch/tools/github/circt/llvm/build/./lib/libMLIRPass.so.15git+0x2b3cb)
#17 0x00007fdcef459f29 mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) (/mnt/scratch/tools/github/circt/llvm/build/./lib/libMLIRPass.so.15git+0x2bf29)
#18 0x00007fdcef45aab0 mlir::detail::OpToOpPassAdaptor::runPipeline(llvm::iterator_range<llvm::pointee_iterator<std::unique_ptr<mlir::Pass, std::default_delete<mlir::Pass> >*, mlir::Pass> >, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) (/mnt/scratch/tools/github/circt/llvm/build/./lib/libMLIRPass.so.15git+0x2cab0)
#19 0x00007fdcef45bdd4 mlir::PassManager::run(mlir::Operation*) (/mnt/scratch/tools/github/circt/llvm/build/./lib/libMLIRPass.so.15git+0x2ddd4)
#20 0x00007fdcf01fe03d performActions(llvm::raw_ostream&, bool, bool, llvm::SourceMgr&, mlir::MLIRContext*, llvm::function_ref<mlir::LogicalResult (mlir::PassManager&)>) (.constprop.0) MlirOptMain.cpp:0:0
#21 0x00007fdcf01fe3a1 processBuffer(llvm::raw_ostream&, std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer> >, bool, bool, bool, bool, llvm::function_ref<mlir::LogicalResult (mlir::PassManager&)>, mlir::DialectRegistry&, llvm::ThreadPool*) MlirOptMain.cpp:0:0
#22 0x00007fdcf01fe850 mlir::MlirOptMain(llvm::raw_ostream&, std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer> >, llvm::function_ref<mlir::LogicalResult (mlir::PassManager&)>, mlir::DialectRegistry&, bool, bool, bool, bool, bool) (/mnt/scratch/tools/github/circt/llvm/build/./lib/libMLIROptLib.so.15git+0x7850)
#23 0x00007fdcf0200622 mlir::MlirOptMain(int, char**, llvm::StringRef, mlir::DialectRegistry&, bool) (/mnt/scratch/tools/github/circt/llvm/build/./lib/libMLIROptLib.so.15git+0x9622)
#24 0x00005589e889ec6c main (/mnt/scratch/tools/github/circt/build/bin/circt-opt+0x4bc6c)
#25 0x00007fdceea21d90 __libc_start_call_main ./csu/../sysdeps/nptl/libc_start_call_main.h:58:16
#26 0x00007fdceea21e40 call_init ./csu/../csu/libc-start.c:128:20
#27 0x00007fdceea21e40 __libc_start_main ./csu/../csu/libc-start.c:379:5
#28 0x00005589e889efa5 _start (/mnt/scratch/tools/github/circt/build/bin/circt-opt+0x4bfa5)
make: *** [/mnt/scratch/tests/makefile:177: polygeist_circt_calyx] Aborted (core dumped)

Source:
[affine.txt](https://github.com/llvm/circt/files/8691291/affine.txt)
[triangle.cpp.txt](https://github.com/llvm/circt/files/8691293/triangle.cpp.txt)

cgyurgyik, 2022-05-14:
Just to be clear the `This is a code snippet` part isn't necessary :-), I was just trying to point out the difference of before and after.

mikeurbach, 2022-05-25:
This looks like there is an assumption on the form of the affine loop, it looks like it needs to handle affine maps with symbols. I will take a look and at least provide a better error message, and hopefully fix this issue using the affine tooling upstream.

mikeurbach, 2022-05-29:
@A-Dane I believe this specific issue should be fixed.

closed 2022-05-29 by mikeurbach

A-Dane, 2022-05-29:
Awesome - Thanks a zillion!