llvm/circt #3139: [Verilator] Verilator 4.222 Shows test/verilator Failures

seldridge, 2022-05-17:
I bumped Verilator locally and noticed two failures in the test/verilator tests.

The first is a change in a warning message. `(=)` is now `'='`:

```
/Users/schuylere/repos/github.com/llvm/circt/test/verilator/errors1.sv:19:11: error: CHECK: expected string not found in input
// CHECK: %Warning-BLKSEQ: {{.*}}:18{{.*}} Blocking assignments (=) in sequential (flop or latch) block
          ^
<stdin>:1:122: note: scanning from here
%Warning-UNUSED: /Users/schuylere/repos/github.com/llvm/circt/test/verilator/errors1.sv:9:14: Signal is not used: 'rst_n'
                                                                                                                         ^
<stdin>:7:67: note: possible intended match here
%Warning-BLKSEQ: /Users/schuylere/repos/github.com/llvm/circt/test/verilator/errors1.sv:18:11: Blocking assignment '=' in sequential logic process
                                                                  ^

Input file: <stdin>
Check file: /Users/schuylere/repos/github.com/llvm/circt/test/verilator/errors1.sv

-dump-input=help explains the following input dump.

Input was:
<<<<<<
            1: %Warning-UNUSED: /Users/schuylere/repos/github.com/llvm/circt/test/verilator/errors1.sv:9:14: Signal is not used: 'rst_n' 
check:19'0                                                                                                                              X error: no match found
            2:  : ... In instance main 
check:19'0     ~~~~~~~~~~~~~~~~~~~~~~~~
            3:  9 | input wire rst_n, 
check:19'0     ~~~~~~~~~~~~~~~~~~~~~~~
            4:  | ^~~~~ 
check:19'0     ~~~~~~~~~
            5:  ... For warning description see https://verilator.org/warn/UNUSED?v=4.222 
check:19'0     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
            6:  ... Use "/* verilator lint_off UNUSED */" and lint_on around source to disable this message. 
check:19'0     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
            7: %Warning-BLKSEQ: /Users/schuylere/repos/github.com/llvm/circt/test/verilator/errors1.sv:18:11: Blocking assignment '=' in sequential logic process 
check:19'0     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
check:19'1                                                                       ?                                                                                 possible intended match
            8:  : ... Suggest using delayed assignment '<=' 
check:19'0     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
            9:  18 | x_int = x_int + 1; 
check:19'0     ~~~~~~~~~~~~~~~~~~~~~~~~~
           10:  | ^ 
check:19'0     ~~~~~
           11: %Error: Exiting due to 2 warning(s) 
check:19'0     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
>>>>>>
```

The second appears to be an actual problem with the Verilog used in the test:

```
: 'RUN: at line 2';   /Users/schuylere/.nix-profile/bin/verilator --cc --top-module main -Wall -Wpedantic /Users/schuylere/repos/github.com/llvm/circt/test/verilator/main.sv
--
Exit Code: 1

Command Output (stderr):
--
%Warning-UNOPTFLAT: /Users/schuylere/repos/github.com/llvm/circt/test/verilator/main.sv:25:15: Signal unoptimizable: Feedback to clock or circular logic: 'main.fooIn'
   25 |   logic [3:0] fooIn = x[5:2];
      |               ^~~~~
                    ... For warning description see https://verilator.org/warn/UNOPTFLAT?v=4.222
                    ... Use "/* verilator lint_off UNOPTFLAT */" and lint_on around source to disable this message.
                    /Users/schuylere/repos/github.com/llvm/circt/test/verilator/main.sv:25:15:      Example path: main.fooIn
                    /Users/schuylere/repos/github.com/llvm/circt/test/verilator/main.sv:15:12:      Example path: ASSIGNW
                    /Users/schuylere/repos/github.com/llvm/circt/test/verilator/main.sv:11:23:      Example path: x
                    /Users/schuylere/repos/github.com/llvm/circt/test/verilator/main.sv:25:24:      Example path: INITIALSTATIC
                    /Users/schuylere/repos/github.com/llvm/circt/test/verilator/main.sv:25:15:      Example path: main.fooIn
%Warning-UNOPTFLAT: /Users/schuylere/repos/github.com/llvm/circt/test/verilator/main.sv:29:15: Signal unoptimizable: Feedback to clock or circular logic: 'main.barIn'
   29 |   logic [9:0] barIn = x[11:2];
      |               ^~~~~
                    /Users/schuylere/repos/github.com/llvm/circt/test/verilator/main.sv:29:15:      Example path: main.barIn
                    /Users/schuylere/repos/github.com/llvm/circt/test/verilator/main.sv:15:12:      Example path: ASSIGNW
                    /Users/schuylere/repos/github.com/llvm/circt/test/verilator/main.sv:11:23:      Example path: x
                    /Users/schuylere/repos/github.com/llvm/circt/test/verilator/main.sv:29:24:      Example path: INITIALSTATIC
                    /Users/schuylere/repos/github.com/llvm/circt/test/verilator/main.sv:29:15:      Example path: main.barIn
%Error: Exiting due to 2 warning(s)

--
```

mikeurbach, 2022-05-17:
These tests appear to essentially test Verilator? It looks to me like they were added to make sure we had a usable Verilator in CI, and we have since moved past this being useful. From what I can see, I'd just remove `test/verilator/`. @teqdruid do you have thoughts here?

teqdruid, 2022-05-17:
Yeah, that was the original intent. I agree that these tests aren't useful anymore, but I do think checking that one has a sane RTL simulator is useful. We don't have any tests specifically for the `circt-rtl-sim` tool, so I think we should simply these tests and convert them to use the `circt-rtl-sim` tool.

mikeurbach, 2022-05-17:
We do have integration tests for circt-rtl-sim, no? I'm thinking of these ones: https://github.com/llvm/circt/blob/main/integration_test/circt-rtl-sim/circt-rtl-sim-verilator.sv. Did you mean non-integration tests, or something else?

mikeurbach, 2022-05-17:
I just double checked, and those integration tests should be enabled as part of the "short integration tests" that run on PRs and post-commit, because Verilator is installed and detected by CMake.

teqdruid, 2022-05-19:
Could have sworn I replied. I'd rather have the circt-rtl-sim tool tested in `test` just to exercise basic functionality, but eh. I've deleted these two tests.

closed 2022-05-19 by teqdruid

teqdruid, 2022-10-11:
Yep, I forgot about those.

I'm OK with deleting the broken tests then.

commit 0caf8adf (2022-05-19): [Tests] Remove Verilator sanity tests

Resolves #3139.