llvm/circt #3175: Version generation fails for out-of-tree builds

teqdruid, 2022-05-24:
```
-- Generating /home/jodemme/circt_build/tools/circt/lib/Support/Version.cpp from /home/jodemme/circt/lib/Support/Version.cpp.in by `git describe --dirty --tags --match circtorg*`
fatal: not a git repository (or any of the parent directories): .git
CMake Warning at /home/jodemme/circt/cmake/modules/GenVersionFile.cmake:21 (message):
  git describe failed, set version to unknown git version
```

I don't build in the CIRCT source tree, so I get the above warning whenever cmake runs.

uenoku, 2022-05-24:
I'll make the version generation opt-in for now.

teqdruid, 2022-05-24:
OK. I wasn't so much worried about the warning. Looks like the cmake file is just using the wrong directory for the .git search.

PR #3199 (2022-05-25): [CMake] Run git describe from the source dir, fix out-of-tree builds
Don't rely on build being same or underneath the source directory.

Fixes #3175.
commit 9f240a83 (2022-05-25): [CMake] Run git describe from the source dir, fix out-of-tree builds (#3199)

Don't rely on build being same or underneath the source directory.

Fixes #3175.
closed 2022-05-25 by dtzSiFive