llvm/circt #3253: [ExportVerilog] Add option to not emit location info

rsetaluri, 2022-06-01:
Sometimes it's preferable to not emit location info (e.g. diffing verilog, code size). We would like an option to disable emitting location info in the Verilog.

rsetaluri, 2022-06-01:
Thanks for the pointers @seldridge! We are not actually going through firrtl/firtool and instead emitting hw/sv directly. Could this option be exposed for the `circt-opt` binary as well?

I'm actually fine being force to emit MLIR that doesn't have location info from the jump -- but it seems like the parser automatically adds it?

seldridge, 2022-06-01:
Yeah, it's reasonable to just register that pass inside `circt-opt` around here: https://github.com/llvm/circt/blob/main/tools/circt-opt/circt-opt.cpp#L55

Noting that it may actually make sense to register this for `circt-translate` as a CLI-based pipeline (e.g., `circt-translate Foo.mlir | circt-opt -strip-debug-info | circt-translate`) will result in the info getting re-processed as "stdin" in the last pipe. There may be some other upstream mechanism to not parse the debuginfo that is more appropriate for `circt-translate`.

uenoku, 2022-06-03:
As another solution, we can add `None` option to `locationInfoStyle`. I think -string-debug-info makes error reporting difficult at ExportVerilog so it would be nice to have `None` option in `locationInfoStyle`.
https://github.com/llvm/circt/blob/96ce99b764b374b241292c9026b6bc834dfbd1a4/docs/VerilogGeneration.md?plain=1#L84-L87

uenoku, 2022-06-24:
Closed by https://github.com/llvm/circt/commit/476b3705444325d5b6b282481c8278223add8464. Thanks @rsetaluri!

closed 2022-06-24 by uenoku