llvm/circt #3256: [StandardToHandshake] Handing unused external memory.

Dinistro, 2022-06-02:
The following IR causes the `--convert-std-to-handshake` pass to produce a segfault.

```mlir
module {
  func.func @foo(%mem : memref<i32>) {
    return
  }
}
```

The cause is `https://github.com/llvm/circt/blob/main/lib/Conversion/StandardToHandshake/StandardToHandshake.cpp#L1457` because there is no such thing as a "defining op".

This can obviously be circumvented by checking if it's a `BlockArgument` and then continuing the loop. While the fix stops the pass from crashing and thus produces the following, the conversion to FIRRTL crashes afterwards.

```mlir
handshake.func @foo(%arg0: memref<i32>, %arg1: none, ...) -> none attributes {argNames = ["in0", "inCtrl"], resNames = ["outCtrl"]} {
  sink %arg0 : memref<i32>
  return %arg1 : none
}

```

`HandshakeToFIRRTL` assumes that each memref which is passed to a function is feed into a `extmemory` operation, which is not the case here. 
How should I fix that? By introducing an extra check that will ignore this case when then `memref` is directly pushed into a sink?

mortbopet, 2022-06-02:
Hmm... one alternative approach to avoid special-case checks everywhere would be to _always_ introduce `extmemory` operations for `memref` block arguments, whether they are referenced by memory accessing operations or not.

We lose nothing in the used-memory case, and in the unused-memory case, I am fairly confident that whatever will be introduced by lowering `extmemory` will be pruned during synthesis.

An alternative is to remove the memref argument from the handshake function, but i would advise against that since it would break the API of the function.

mikeurbach, 2022-06-02:
> An alternative is to remove the memref argument from the handshake function, but i would advise against that since it would break the API of the function.

That seems a little risky as a compiler transform, but is probably what the frontend generating this IR should do :).

Lowering to an `extmemory` that is unused could be one approach, but what about just emitting an error message? It's easy to write this kind of IR by hand, but if it's produced by a machine wouldn't it always be an error? Or if not always an error, produce a warning and then lower it to an unused `extmemory`.

mikeurbach, 2022-06-02:
Regardless, thanks for raising the issue, the compiler shouldn't crash on valid input.

If we wanted this to always be an error, it could be a verifier rule.

mortbopet, 2022-06-02:
I'd support a warning + unused `extmemory` but not a big fan of erroring out on incoming `func.func`-level code with unused memref arguments. In that case, I think it's more valuable to have code passing through the compiler cleanly than to error out. Regardless, the `handshake.func` verifier should probably be extended to ensure that `memref` arguments are referenced exactly once by a `handshake.extmemory` in its body.

Dinistro, 2022-06-03:
Thanks for clarifying this. I'll work on a fix for that once I'm done with some other bugs that I have to hunt down.

mortbopet, 2022-06-03:
Much appreciated, and thank you for your work here - debugging Handshake IR and the hardware it generates is non-trivial!

PR #3266 (2022-06-05): [StandardToHandshake] Always emit `extmemory`, even when unused
This PR ensures that all memrefs are connected to memory by initializing the `memRefOps` map with an empty vector for all block arguments. The map is later on used to produce `ExternalMemrefOps`. 

Fixes #3256
closed 2022-06-05 by Dinistro