llvm/circt #3273: [FIRParser] unexpectedly accepting of invalid FIRRTL

dtzSiFive, 2022-06-03:
This started with a module definition at the wrong indentation level-- we currently we accept "FIRRTL" like this:

```scala
circuit Oops:
  module Oops:
  input in: UInt
  input out: UInt
  wire w: UInt
  w <- in
  out <- w
```

Producing an empty module "Oops" and discarding the statements.

Poking at things, it seems we even accept things like:

```scala
circuit Oops:
  module Oops:
    input in: UInt<1>
    output out: UInt<1>
    out <- in

  asdf:
    dsfa

asdf
input in: UInt
  input out: UInt
  wire w: UInt
  w <- in
  out <- w
  input in: UInt
```

seldridge, 2022-06-04:
Yeah, this is wrong.  This is one area where the FIRRTL spec actually has a lot of clarifying information on how this is supposed to work (which is admittedly based on how the SFC works).

dtzSiFive, 2022-06-08:
Looks like we call success when parsing statements once the indent level drops too low (below level started at), and since we mostly-blindly chunk the input by "module"/EOF (or so, for parallel processing), this combines to result in anything at indent level <= 1 before the next 'module' or EOF being allowed (as seen in example above, once indent level drops anything goes).

I haven't looked into where it might make sense to check for these sorts of unprocessed bits (or otherwise how best to address this).
I'll take a look at that part of FIRRTL spec, thanks!

This also explains behavior encountered in #2978 (second circuit's module parsed into first circuit without a warning).

dtzSiFive, 2022-09-21:
Previously: #2787.