llvm/circt #3328: [CI] Nightly integration tests starting to fail due to disk space issues

teqdruid, 2022-06-11:
Some of the nightly integration tests are starting to fail due to disk space issues. (The GCC debug -- no surprise there.) MLIR compiles are too large IMO. Are there some cmake flags or changes we can make to avoid pulling in as much LLVM as we do? FFS, I've seen OpenMP stuff being pulled in!

mikeurbach, 2022-06-11:
Some low hanging fruit there is sometimes `-DLLVM_TARGETS_TO_BUILD="host"`, which appears to be supplied already.

I am trying to remember the situation that includes all the cruft like OpenMP.

I'm not sure if this is _the_ culprit, but I recall there is something about how the Python bindings depend on MLIR that we can probably fix. IIRC we were updating in lockstep with a few changes upstream, but lagged behind one final change that I should have done but dropped the ball on, which could help here. I'm not sure how significant it will be, but since we're building Python nightly it will help.

mikeurbach, 2022-06-11:
I found what I think I was looking for posted Discord almost a year ago: https://discord.com/channels/636084430946959380/742572728787402763/870051970315075655.

Copied here:

```
@mikeurbach Once https://reviews.llvm.org/D106992 lands, if you change your python API declarations, you'll get about a 50% savings in files to build (and binary size):

set(_source_groups
  MLIRPythonSources.Core
  MLIRPythonSources.Dialects
  MLIRHLOPythonSources
  MLIRHLOPythonExtensions
)

add_mlir_python_common_capi_library(MLIRHLOCAPI
  INSTALL_COMPONENT MLIRHLOPythonModules
  INSTALL_DESTINATION python_packages/mlir_hlo/mlir/_mlir_libs
  OUTPUT_DIRECTORY "${MLIR_HLO_BINARY_DIR}/python_packages/mlir_hlo/mlir/_mlir_libs"
  RELATIVE_INSTALL_ROOT "../../../.."
  DECLARED_SOURCES ${_source_groups}
)

add_mlir_python_modules(MLIRHLOPythonModules
  ROOT_PREFIX "${MLIR_HLO_BINARY_DIR}/python_packages/mlir_hlo/mlir"
  INSTALL_PREFIX "python_packages/mlir_hlo/mlir"
  DECLARED_SOURCES ${_source_groups}
  COMMON_CAPI_LINK_LIBS
    MLIRHLOCAPI
  )
^---- An example of a pending change doing similar things for mlir_hlo.
There is more that we can do to disaggregate the dependency on the core dialects, which accounts for the bulk of the rest of the build, but I'm going to take a breather for now as where we're at is good enough for me (for now).
If you just do ninja install-MLIRHLOPythonModules (or use directory based :all rules), it will build/install a lot less (vs ninja).
```

Looking at what we have, I think we already slimmed this down to just `MLIRPythonSources.Core`, so I would hope OpenMP isn't coming from there: https://github.com/llvm/circt/blob/2174fb368c6d378d5e43805561bed883e5a6b360/lib/Bindings/Python/CMakeLists.txt#L128-L131

Also for the record the thing we didn't get to with Python bindings is about namespacing, which I also noticed on discord.

mikeurbach, 2022-06-12:
Ok one more small data point here: I just did a fresh build of the `circt-opt` target after pulling llvm, which did not pull in all that openmp stuff. Then I built the `check-pycde` target, which did include all that. I havent't pinned it down yet, but I think something about the Python bindings still pulls in more than necessary.

mikeurbach, 2022-06-12:
While I was waiting for the pycde build, I took a look, and I think this is what I've was thinking of: https://github.com/llvm/llvm-project/blob/571ae1abebb6b102cc2e95e3abf8e81757fba35a/mlir/python/CMakeLists.txt#L279

I think that target is what is effectively pulling in a huge chunk of MLIR: https://github.com/llvm/llvm-project/blob/571ae1abebb6b102cc2e95e3abf8e81757fba35a/mlir/lib/CAPI/Registration/CMakeLists.txt#L2-L13

I can try to disaggregate that and if it is working well send a patch

mikeurbach, 2022-07-06:
I finally got around to testing the "disaggregate" patch (https://reviews.llvm.org/D128593). It looks like that will reduce the size of CIRCT's Python shared library from 379 MB to 84 MB. The only change I had to do to incorporate that was remove all includes of `mlir-c/Registration.h` (and replace it with includes of `mlir-c/IR.h` if not already included).

So, we can enjoy close to 300 MB of savings as soon as that patch is landed and incorporated to CIRCT. I'll be monitoring that this week--for other reasons I plan to bump LLVM to the latest this week anyway.

teqdruid, 2022-07-06:
Woot woot!

teqdruid, 2022-07-06:
I’m also assuming that this patch drastically reduced the number of files which are getting compiled? (And thus solve the disk space issue we are running into.)

mikeurbach, 2022-07-06:
On a clean build, with this change, I saw it compiling around ~1100 files for the CIRCTMLIRPythonModules target. Before the change, I didn't do a proper clean build, but the machine hadn't updated CIRCT/LLVM/MLIR in months, and I pulled LLVM fresh, so it was probably pretty close to a full re-compile, and that brought in ~2400 files. So yes, more than 50% reduction in number of files getting compiled.

teqdruid, 2022-07-19:
So that submodule bump didn't do it since the CI flow is still compiling a comparable number of files (3286 vs 3289). That's pretty similar to what my system compiles.

teqdruid, 2022-07-19:
Yep.

Is it possible to not pull in as much for `circt-opt`? I'm seeing like the `amdgpu` and `openacc` dialects get pulled in, which presumably pull in LLVM support for GPUs, etc. Does MLIR allow this level of dependence selection? (and work?)

mikeurbach, 2022-07-19:
I'm not sure where those come from... I wonder if there is a way to trace transitive dependences in cmake. We should be able to depend on what we need, and I don't think we need those dialects at all.

teqdruid, 2022-07-19:
https://cmake.org/cmake/help/latest/module/CMakeGraphVizOptions.html ?

teqdruid, 2022-09-23:
the nightlies are failing again due to disk space issues. As usual, it's gcc debug. At this point, I think we should just disable all of the gcc debug builds. What say y'all?

mikeurbach, 2022-09-26:
I would vote yes, let's temporarily remove gcc debug builds. It seems more important to have working integration tests than that corner of the build matrix.

We should also revive efforts to slim things down. It makes no sense to be compiling all that crap we don't care about like amdgpu and openacc.

closed 2022-09-26 by teqdruid

mikeurbach, 2022-09-29:
Thanks @teqdruid!

PR #3999 (2022-09-26): [CI] Disable gcc/debug on nightly tests
GCC debug binaries are just too large and have been causing disk space issues intermittently. Just disable them for now as it is a longer effort to cut down on the amount of stuff we build. Fixes #3328 .
commit e5bb57eb (2022-09-26): [CI] Disable gcc-debug (#3999)

GCC debug binaries are just too large and have been causing disk space issues intermittently. Just disable them for now as it is a longer effort to cut down on the amount of stuff we build. Fixes #3328 .