llvm/circt #3329: [FIRRTL][Verilog] Use Xcelium Coverage Exclusion Magic in Addition to VCS

seldridge, 2022-06-11:
Currently we only emit extracted covers with the VCS magic file-level comment to exclude them:

```
// VCS coverage exclude file 
...
```

However, anywhere we emit this, we should also emit the Xcelium magic off/on comments. Xcelium looks like:

```
// pragma coverage block = off, toggle = off
...
// pragma coverage block = on, toggle = on
```

I am not super stoked about this option (and also note that the Xcelium pragmas I've been given are turning off specific types of coverage which may not be the case for all users 🥲 ) and would like to have a better mechanism to indicate that something is a cover, what file it should go to, and what simulators you are using so that we aren't just dropping verbatim Verilog comments all over the place.

darthscsi, 2022-06-11:
We need an encoding for the behavior and lowering/emission options for different (possibly multiple) tools.

uenoku, 2022-06-11:
>  would like to have a better mechanism to indicate that something is a cover, what file it should go to

+1 on this. I'm wondering if we can create `coverage_exclude_file.f` so that we can exclude those files directly from tools instead of relying on pragmas.  Implementation wise, I think we can add "coverage_exclude_module" attribute to modules we want to exclude and gather them in ExportVerilog.