llvm/circt #3363: [FIRParser] Failure to parse instance named "module"

tymcauley, 2022-06-15:
The [Gemmini](https://github.com/ucb-bar/gemmini) design contains a module instance named "module": https://github.com/ucb-bar/gemmini/blob/87550581d98210540ac013f96fbcb9150be39df7/src/main/scala/gemmini/CounterFile.scala#L228

This results in the following parser failure:

```
$ ./build/bin/firtool --format=fir --warn-on-unprocessed-annotations --verify-each=false --dedup=1 --annotation-file=chipyard.TestHarness.GemminiRocketConfig.anno.json -o Gemmini.v chipyard.TestHarness.GemminiRocketConfig.fir --verbose-pass-executions
[firtool] Running fir parser
chipyard.TestHarness.GemminiRocketConfig.fir:285245:10: error: 'module' must be first token on its line
    inst module of CounterFile @[CounterFile.scala 228:24]
         ^
```

It looks like while parsing the contents of a module, we immediately exit the module-statement-parser when we encounter `module`/`extmodule`: https://github.com/llvm/circt/blob/23ae1a7391c3d4d03eceade2189e224c4bb1ebaf/lib/Dialect/FIRRTL/Import/FIRParser.cpp#L3577-L3580

We then pop back out to the circuit parser, and try to interpret that `module` token as the start of a module definition, and then trigger this error: https://github.com/llvm/circt/blob/23ae1a7391c3d4d03eceade2189e224c4bb1ebaf/lib/Dialect/FIRRTL/Import/FIRParser.cpp#L3826-L3830

It appears this strategy is the problem: https://github.com/llvm/circt/blob/23ae1a7391c3d4d03eceade2189e224c4bb1ebaf/lib/Dialect/FIRRTL/Import/FIRParser.cpp#L3567-L3568

Since we can't determine when we're in the middle of lexing a module statement, and therefore if this `module` token is in the middle of a statement, or at the start of a new one.

tymcauley, 2023-03-23:
We can close now that #4589 is merged, right?

adkian-sifive, 2023-06-21:
Fixed by #4589

closed 2023-06-21 by adkian-sifive