llvm/circt #3417: [FIRRTL] Unused 0-bit firrtl.invalidvalue remains after LowerToHW

uenoku, 2022-06-24:
Input:
```mlir
firrtl.circuit "Foo"  {
  firrtl.module @Foo() {
    %invalid_ui0 = firrtl.invalidvalue : !firrtl.uint<0>
  }
}
```
Output of `circt-opt -lower-firrtl-to-hw` contains `firrtl.invalidvalue` for some reason.
```
  hw.module @Foo() {
    %invalid_ui0 = firrtl.invalidvalue : !firrtl.uint<0>
    hw.output
  }
```

seldridge, 2022-06-27:
Two things:

(1) Whatever is causing these to hit `LowerToHW` should be investigated as an SFC-compliant pass pipeline shouldn't be allowing these to live beyond `SFCCompat`. (Perhaps there should be an option to warn/error if running `SFCCompat`?)
(2) `LowerToHW` should do something with them. I think it used to convert these to zero at some time (which is reasonable to be SFC compatible). It is probably better to lower these to `x` and make this a hard-error for our internal flow.

uenoku, 2022-06-28:
> (1) Whatever is causing these to hit LowerToHW should be investigated as an SFC-compliant pass pipeline shouldn't be allowing these to live beyond SFCCompat. (Perhaps there should be an option to warn/error if running SFCCompat?)

Yes, I think dead invalid values are not deleted by SFCCompat pass.

> (2) LowerToHW should do something with them. I think it used to convert these to zero at some time (which is reasonable to be SFC compatible). It is probably better to lower these to x and make this a hard-error for our internal flow.

Actually I guess this bug is in the special handling of zero-bit width in LowerToHW because `hw.constant 0 : i0` is not allowed now.

uenoku, 2022-06-28:
For (1), dead invalids are erased by https://github.com/llvm/circt/commit/632e6e1b0e37e95568ccd9607e76ee8e4aaf6b35

seldridge, 2023-07-30:
Closing this since the original issue is fixed.

closed 2023-07-30 by seldridge