llvm/circt #3429: [StandardToHandshake] Add transformation for feed-forward task pipelining

Dinistro, 2022-06-28:
I'm working on introducing transformation for safe task pipelining in feedforward computational DAGs. This issue should serve as a place to document the progress and discuss the different steps required for this feature. 

@mortbopet Already implemented a task pipelining "guard" for loops that ensures that only one control token is live in a loop. This ensures a correct execution, as it prevents collisions of control tokens. In his [Master's Thesis](https://infoscience.epfl.ch/record/292189?ln=en) it is additionally described how a similar transformation for feed-forward computational graphs could be implemented but it was omitted due to it not being required for his use cases. In my stream project, the generated circuits are continuously fed with new control tokens. Therefore, I observed reordering due to having different branches with different latencies. 

As of yet, I do have test cases that demonstrate the reordering of results due to inputs taking different paths. To simulate this, I created a new test driver in SystemVerilog that feeds multiple inputs to one circuit. It requires some cleanup but should be usable for other test cases. 
Furthermore, I started to implement a transformation that inserts new blocks for all conversion points. Having these blocks already beforehand simplifies the transformations substantially (Figure 6.10 in the thesis). 

As of now, I do see the following steps:
- [ ] Add test driver to test circuits with multiple inputs. This can be a separate PR I think.
- [ ] Introduce a transformation that inserts explicit merge blocks where needed. Can be a separate PR as well.
- [ ] `FeedForwardNetworkRewriter` class, similar to `LoopNetworkRewriter`, which inserts the necessary logic.

mortbopet, 2022-06-28:
Looking forward to seeing how this turns out!

wrt. the `FeedForwardNetworkRewriter`, since this will be at the `cf` level, i suspect this will be suitable for upstreaming into MLIR. The philosophy of this kind of CFG transformation for dataflow purposes is much like the max-ssa pass (which i've neglected following up on to get upstreamed... https://reviews.llvm.org/D114945).

Dinistro, 2022-06-28:
I'm not sure if we can upstream this to MLIR at some point as this transformation only works for certain cases. One case that is a bit odd to handle is the following:

[image]

My current idea is to simply only allow one thread in such a construct, similar to loops, as other solutions might be very complicated.

mortbopet, 2022-06-28:
Hmm yea that is a bit of a different case than what i presented in the thesis. What can and can't be merged rewritten should probably be more formally defined (maybe some petri net/marked graph theory could apply). In my mind i view each split as giving downstream control flow nodes a new "color" - and once identically colored nodes meet, a merge node can be inserted. This is therefore not the case for the example you showed:
[image]

In a sense, differently colored nodes converging implies a structural conflict - two independent threads require use of the same resource at once. An as such, to remove that structural conflict, we need to either do operator sharing (that is, only allowing one thread in the code) or increase the resources available in the program - which, at the CFG level, would imply that we **copy** the subset of the CFG which has a structural conflict:

[image]

Dinistro, 2022-06-28:
Thanks for sharing your idea together with these helpful visualizations. 

I'm currently trying to formalize something similar that remembers the last dominator with multiple successors. I'll need some more time to finish this though.

Dinistro, 2022-07-01:
As a small note: Tobias just pointed me to single-entry-single-exit regions in LLVM that might be comparable to what we need for the the insertion of the guards: https://github.com/llvm/llvm-project/blob/main/llvm/include/llvm/Analysis/RegionInfo.h#L191

closed 2022-08-10 by Dinistro