llvm/circt #3474: Cap'n Proto And Fibers / Exceptions

seldridge, 2022-07-05:
I did a bump of system-installed packages and there may be an issue with some distributions of Cap'n Proto for versions which have fibers enabled. It looks like fibers are incompatible with no-exception build options. E.g., I started to see the following build failure on a rebuild of CIRCT:

```
/nix/store/4w0pgdp0nv391sksvi179zjq3x7wvifz-capnproto-0.10.1/include/kj/async.h:39:6: error: "Fibers cannot be enabled when exceptions are disabled."
    #error "Fibers cannot be enabled when exceptions are disabled."
     ^
1 error generated.
```

This is probably a @teqdruid question on how to handle this.

I can always disable Cap'n Proto in my build.

teqdruid, 2022-07-06:
You should probably disable capnp builds until I have a chance to dig into this. The part of capnp which is used by CIRCT shouldn’t need fibers, so this should be solveable.

teqdruid, 2022-07-08:
@seldridge I haven't been able to repro this. Can you post your nix file?

seldridge, 2022-07-08:
I'm not doing anything that fancy (though I do not fully recall what my setup is here). I'm just using `nix-env` instead of homebrew on a macOS machine. What I think I would test is nothing to do with nix, but if CIRCT works with Cap'n Proto >=0.10.0. This check was added seemingly in this commit (https://github.com/capnproto/capnproto/commit/f0d37520365c9bbc343a631fe732dc6504a6dd36) which is in >=0.10.0.

This likely means that Cap'n Proto needs to be built with `KJ_USE_FIBERS=0` for it to work with CIRCT.

teqdruid, 2022-07-08:
Yeah, I'm compiling capnp 0.10.1 using `utils/get-capnp.sh` (with `--with-fibers` added) and for some reason I can't get that error to fire.

Can you send over the compile log? There are two places capnp is used -- one of which should be getting compiled with exceptions -- so I'm wondering which one is tripping this error.

dtzSiFive, 2022-07-20:
Not sure what CIRCT can do to help here, hopefully not a pain point elsewhere too.  But I can help with nix to workaround if that's useful:

FWIW, installing build deps into a user profile isn't always expected to work, but that's not the issue here.

You can override the package to disable fibers if you'd like, few options there.

Overlays is probably the way to go, either drop an appropriate overlay into (the OSX equivalent?) of `~/.config/nixpkgs/overlays.nix` (will be used by nix-env, but also any packages that depend on it will use the overridden version, etc.-- which generally is a feature of overlays, creating consistent package sets) or for more explicit control or a one-off just do something like this and install from there:

`nofibers.nix`:
```nix
import <nixpkgs> {
  overlays = [
    (self: super: {
      capnproto = super.capnproto.overrideAttrs(o: {
        cmakeFlags = o.cmakeFlags or [] ++ [ "-DWITH_FIBERS=OFF" ];
      });
    })
  ];
}
```

And then install/upgrade with:
```
$ nix-env -f nofibers.nix -uA capnproto --leq
```
You'll need to use this when upgrading in the future, something `nix profile` fixes re:tracking installation sources but don't recommend that yet.

(Ran into same issue building circt against Nixpkgs's capnproto this morning, but I have an expression (flake) driving the build so the override is done there.  Haven't tried to reproduce otherwise.).

Or, yeah, disable it in the build for now :smile: .

teqdruid, 2022-07-20:
@dtzSiFive Can you include a build log of the failure? I still can't repro this issue.

dtzSiFive, 2022-07-20:
> @dtzSiFive Can you include a build log of the failure? I still can't repro this issue.

Sure thing!

https://gist.github.com/dtzSiFive/95b17a653ca0bc6b46ba04c64a94b4b0 , error is at the end but posted the full log JIC.

PR #3576 (2022-07-27): [ESI] [Capnp] Compile library with exceptions
The EsiCosimCapnp library never gets linked into any MLIR code. It is
used exclusively to build the DPI shared library. As such, it can get
compiled with exceptions.

Should fix #3474 .
closed 2022-07-27 by teqdruid