llvm/circt #3505: [FIRRTL][LowerTypes] DontTouch on an instance port moves the dont touch on to the instance

youngar, 2022-07-11:
I am mostly curious if this is what is supposed to happen, and there doesn't appear to be any tests of this behvaiour.

A don't touch on an instance's port is moved to the instance itself during LowerTypes:
```
firrtl.circuit "Test" {
  firrtl.module @Test() {
    %foo_x = firrtl.instance foo @Foo(in x : !firrtl.bundle<y : uint<1>> [{circt.fieldID = 1, class = "firrtl.transforms.DontTouchAnnotation"}])
  }
  firrtl.module @Foo(in %x : !firrtl.bundle<y : uint<1>>) { }
}
```

`/bin/circt-opt --firrtl-lower-types ./test3.mlir`
```mlir
module {
  firrtl.circuit "Test"  {
    firrtl.module @Test() {
      %foo_x_y = firrtl.instance foo sym @symfoo  @Foo(in x_y: !firrtl.uint<1>)
    }
    firrtl.module @Foo(in %x_y: !firrtl.uint<1>) {
    }
  }
}
```

youngar, 2022-07-12:
From an offline message, @seldridge notes that this code path doesn't get exercised in practice because the Instance op annotations are moved onto the Module as non-local annotations:
```
circuit Foo: %[[
 {"class":"circt.test", "target":"~Foo|Foo>bar.a"}
]]
  module Bar:
    input a: UInt<1>
  module Foo:

    inst bar of Bar
    bar.a is invalid
```
Becomes:
```mlir
  firrtl.circuit "Foo"  {
    firrtl.hierpath @nla [@Foo::@bar, @Bar]
    firrtl.module private @Bar(in %a: !firrtl.uint<1> [{circt.nonlocal = @nla, class = "circt.test"}]) {
    }
    firrtl.module @Foo() {
      %bar_a = firrtl.instance bar sym @bar interesting_name  @Bar(in a: !firrtl.uint<1>)
      %invalid_ui1 = firrtl.invalidvalue : !firrtl.uint<1>
      firrtl.strictconnect %bar_a, %invalid_ui1 : !firrtl.uint<1>
    }
  }
```

seldridge, 2022-07-12:
Noting that this is true only in the LowerAnnotations path which is the only way after https://github.com/llvm/circt/pull/3497 lands. This is still reachable while FIRAnnotations.cpp still exists.

youngar, 2023-04-26:
Closing since we switched to LowerAnnotations, which always move annotations on InstanceOps to ModuleOps.