llvm/circt #3515: [firtool] -verbose-pass-executions prints wrong options for drop-names

dtzSiFive, 2022-07-12:
Specifying various `-preserve-values=` options (default is 'none' presently) does change the option but not as expected:

```
$ ./bin/firtool ./test.fir -verbose-pass-executions -preserve-values=none |& grep -o "drop-names{preserve-values=[^}]\+}"
drop-names{preserve-values=named}
drop-names{preserve-values=named}
drop-names{preserve-values=named}
drop-names{preserve-values=named}
```

none is default, and is reported as 'named' (shown above).  Here's what's printed for each of the supported values:

"none" -> "named"
"named" -> "none"
"all" -> "none"

AFAICT the options are working properly, just printed/reported incorrectly.

uenoku, 2022-07-12:
Thanks, I'll take a look. The printer simply relies on `printAsTextualPipeline` method so it might be a MLIR problem.  

https://github.com/llvm/circt/blob/f6f2492e3c582eb9d6505724e2c25f25f1154add/tools/firtool/firtool.cpp#L375

dtzSiFive, 2023-09-13:
This works now after latest LLVM bump!

Checked after seeing related upstream fix for `cl::opt` equality: https://github.com/llvm/llvm-project/pull/65754 :tada: .

Woohoo!

closed 2023-09-13 by dtzSiFive