llvm/circt #3539: [LLHDToLLVM] Refactor conversion patterns of each dialect to separate files

maerhart, 2022-07-15:
There is a lowering pass called LLHDToLLVM (CLI flag `--convert-llhd-to-llvm`) that collects all conversion patterns necessary to lower a design compiled by the Moore compiler (which outputs `comb`, `hw`, and `llhd` operations) to LLVM IR code that can be used by llhd-sim for simulation. Most of those patterns are more general though and could be useful for other backends. This includes all `comb` and `hw` patterns. Therefore, it would be a good idea to separate them in a library that allows to only run the lowering for one of those dialects or populate a new lowering pass with the patterns of just one dialect from the library, possibly provide custom replacements for some patterns, and add patterns for other operations to that new pass.

This could look like the following:
- [ ] Create a file like [`LLHDToLLVM.h`](https://github.com/llvm/circt/blob/main/include/circt/Conversion/LLHDToLLVM.h) for each dialect, i.e., `HWToLLVM.h` and `CombToLLVM.h`
- [ ] Provide a file like [`LLHDToLLVM.cpp`](https://github.com/llvm/circt/blob/main/lib/Conversion/LLHDToLLVM/LLHDToLLVM.cpp) for `comb` and `hw` and move the conversion patterns specific to these dialects there and register them in the `populateCombToLLVMConversionPatterns` and `populateHWToLLVMConversionPatterns` functions. Then use them to expose a conversion pass just like `--convert-llhd-to-llvm` with the difference that they only lower `comb` or `hw` ops respectively. If the IR contains ops of other dialects, the pass should insert `UnrealizedConversionOp` to convert the types of the values. Thus it has to be marked legal as, e.g., [here](https://github.com/llvm/circt/blob/main/lib/Conversion/LLHDToLLVM/LLHDToLLVM.cpp#L2695). This allows to call `circt-opt --convert-hw-to-llvm --convert-comb-to-llvm --reconcile-unrealized-casts` to completely convert a design that contains only `hw` and `comb` operations instead of writing a separate pass that populates the conversion patterns of both.
- [ ] Let the LLHDToLLVM conversion pass depend on the other two and call the populate functions to add the conversion patterns.

Stretch goal: allow to specify a subset of operations for which the conversion patterns should be populated (e.g., only `comb.mux` and `comb.extract` instead of all `comb` operations). This could be useful if you want to reuse most conversions in the library, but just have that one operation for which you want to do something specific and thus want to provide a conversion pattern on your own. The call could then look like `populateCombToLLVMOpPatterns<comb::MuxOp, comb::ExtractOp>(patterns);` and likely requires some boilerplate code to map the provided operation template parameters to the conversion patterns. We don't want to make the patterns directly accessible outside of the files they are defined in.

TaoBi22, 2022-07-15:
Happy to work on this!

closed 2022-10-11 by maerhart

tobiasgrosser, 2022-10-11:
Excellent. Thank you Martin.

maerhart, 2022-10-11:
> Excellent. Thank you Martin.

@TaoBi22 did all the work here. Just noticed that this hasn't been closed yet. Thanks Bea!