llvm/circt #3549: [HWArith] Implement remaining operators

mortbopet, 2022-07-18:
This issue serves as a tracking commit for implementing the remaining operations in `hwarith`

- [ ] `hwarith.shl`
- [ ] `hwarith.shr` (sign and zero extending)
- [ ] Comparison operators `<,>,==,<=,>=`.
- [ ] Casting (signed <-> unsigned, truncation/extension)

teqdruid, 2022-07-18:
For shr and shl, how would they differ from the existing comb operators?

jopperm, 2022-07-20:
> For shr and shl, how would they differ from the existing comb operators?

I think the main reason for having `hwarith` shifts is that the `comb` operators don't accept the `ui`/`si` types, so in order to use them inside `comb` land, you'd have to cast operands and results all the time.

Also, in `hwarith`, we only need one op for the right shift, the operands may have different bitwidths, and we have to discuss what happens in case the shift amount is a negative value.

mortbopet, 2022-08-01:
Regarding negative (signed) shift amount operands, I would approach this in a pragmatic way. Unless/until someone find an issue with this, I’d take the approach of Chisel, wherein the shift amount must be an unsigned value – simplifies rules, understanding, and lowering. If shifting with a signed shift amount is needed, I’d expect a user (or a front-end) to decide the semantics of that.