llvm/circt #3574: [FIRRTL] Use EnumAttr for `sym_visibility` of `InnerSymPropertiesAttr`

uenoku, 2022-07-20:
Currently `sym_visibility` uses `StringAttr` as a container but `sym_visibility` is either `public` or `private`. Therefore, it would be cleaner to use enum attr to represent visibility.  
https://github.com/llvm/circt/blob/04797976d2ccdb6e3d1f8ecdc9dee60cc3d5a8c5/include/circt/Dialect/FIRRTL/FIRRTLAttributes.td#L203-L208

shourya5, 2022-07-21:
Just to ensure I understand this issue correctly, "::mlir::StringAttr" in line 208 must be replaced with "::mlir::EnumAttr" right?

uenoku, 2022-07-22:
Thank you for the interest!

> Just to ensure I understand this issue correctly, "::mlir::StringAttr" in line 208 must be replaced with "::mlir::EnumAttr" right?

Yes. On the second thought, I found that MLIR visibility also uses StringAttr (https://github.com/llvm/llvm-project/blob/f359eac5df06c1062019fad9aac41f4320899c5f/mlir/lib/IR/SymbolTable.cpp#L218-L230) for visibility so we might want to keep this convention ... (or need to modify MLIR side as well for consistency). Therefore this issue might be more complicated than what I originally expected so I've removed "good first issue" (contribution is welcome though!)