llvm/circt #3581: [PyCDE] Enable linting

teqdruid, 2022-07-21:
Requires generating python headers for pybind11. MLIR seems to do this (or my editor has some other way of discovering).

mikeurbach, 2022-07-21:
Are you referring to linting or type checking? If type checking, MLIR does maintain stubs, and it would be great to have for CIRCT Python bindings and PyCDE as well. If you meant linting like with flake8 or something, I will open a separate issue about type checking.

teqdruid, 2022-07-21:
I meant type checking. Don't we already have flake8-style checks?

mikeurbach, 2022-07-21:
> Don't we already have flake8-style checks?

Oh yes, we do have that.

I'm happy to help with this. I've been looking into mypy and how MLIR is set up. We will need to generate and maintain stubs for the pybind11 generated parts, and should incrementally add type annotations to our own Python code.

teqdruid, 2022-07-22:
> We will need to generate and maintain stubs for the pybind11 generated parts

I know there's a way to generate them from pybind11 so I'm hoping it's relatively easy to add some cmake magic to generate them. What do you mean by 'maintain'? I had hoped there wouldn't be manual intervention.

mikeurbach, 2022-07-25:
>  What do you mean by 'maintain'?

If you change a pybind11 API, I think there will have to be some step to re-generate and check in the stubs, or else they'll be out of date. If that can be automated by cmake that's great.