llvm/circt #3634: [hwarith] [Python] Implement bitwidth rules in Python

teqdruid, 2022-07-31:
Current in the Python bindings for hwarith `ui8 + ui8 = ui8` which doesn't match the op rules.

We have two options for computing them: (1) re-implement in python or (2) add a CAPI call into CIRCT to use the same bitwidth computation code. The former is the way we traditionally do this (e.g. slicing ops). It occurs to me, however, that a somewhat more generic mechanism could be helpful.

teqdruid, 2022-07-31:
I appear to be wrong about this. I can't find where its implemented, but a pycde test program indeed has the correct bitwidth for mul.

closed 2022-07-31 by teqdruid

mortbopet, 2022-07-31:
These are implicitly implemented by the HWArith operation builders - PyCDE will go create an HWArith operation, that operation only requires the operands to be defined (result type is inferred). Operation is created through CAPI and PyCDE will wrap whatever return value it inspects the newly created operation to have (which has an inferred type).