llvm/circt #3635: [hwarith] Failed to materialize error with ui8 module output

teqdruid, 2022-07-31:
```
hw.module @test1() -> (out: ui8) {
  %1 = hwarith.constant 1 : ui8
  hw.output %1 : ui8
}
```

yields

```
hwarith_test.mlir:2:8: error: failed to materialize conversion for result #0 of operation 'hwarith.constant' that remained live after conversion
  %1 = hwarith.constant 1 : ui8
       ^
hwarith_test.mlir:2:8: note: see current operation: %1 = "hwarith.constant"() {rawValue = 1 : ui8} : () -> ui8
hwarith_test.mlir:3:3: note: see existing live user here: hw.output %0 : ui8
  hw.output %1 : ui8
```

teqdruid, 2022-07-31:
I've also observed this with `hwarith.cast`.

mortbopet, 2022-07-31:
I suspect this is due to `hwarith-to-hw` missing type materialization patterns which will also rewrite `HWModuleOp,HWInstanceOp`.

closed 2022-08-03 by mortbopet

teqdruid, 2022-08-03:
Thanks for fixing this!

PR #3643 (2022-08-03): [HWArithToHW] Add type conversions for 'other' ops
Fixes #3635. Adds signature conversion for `hw::HWModuleOp,hw:HWExternalModuleOp` (`mlir::populateFunctionOpInterfaceTypeConversionPattern` unfortunately can't be provided an interface, i.e. `HWModuleLike`) as well as conversions for 'other' operations which we expect to be mixed in with `hwarith` operations.

Let me know if I've missed any ops for the `ArgResOpConversion` pattern. Any operation which we expect to be used in conjunction with the `ui/si` types of the `hwarith` operations must (unfortunately) be explicitly mentioned.
commit b003931d (2022-08-03): [HWArithToHW] Add type conversions for 'other' ops (#3643)

Fixes #3635. Adds signature conversion for `hw::HWModuleOp,hw:HWExternalModuleOp` (`mlir::populateFunctionOpInterfaceTypeConversionPattern` unfortunately can't be provided an interface, i.e. `HWModuleLike`) as well as conversions for 'other' operations which we expect to be mixed in with `hwarith` operations.