llvm/circt #3641: [FIRRTL] Improve emission of aggregate mux

uenoku, 2022-08-01:
```scala
circuit Foo:
  module Foo:
    input in1: UInt<2>[2]
    input in2: UInt<2>[2]
    input cond: UInt<1>
    output o: UInt<2>[2]
    when cond:
      o <= in1
    else:
      o <= in2
```

The current output with `firtool -preserve-aggregate=1dvec -preserve-public-types=false -firrtl-check-comb-cycles` is:
```
module Foo(
  input  [1:0][1:0] in1,
                    in2,
  input             cond,
  output [1:0][1:0] o);

  wire [1:0][1:0] _o_output;

  assign _o_output[1'h0] = cond ? in1[1'h0] : in2[1'h0];
  assign _o_output[1'h1] = cond ? in1[1'h1] : in2[1'h1];
  assign o = _o_output;
endmodule
```
Optimal result is:
```
module Foo(
  input  [1:0][1:0] in1,
                    in2,
  input             cond,
  output [1:0][1:0] o);

  assign o = cond ? in1 : in2;
endmodule
```
We can implement this transformation within MergeConnection pass.

dtzSiFive, 2023-10-04:
Running this input with: `firtool -preserve-aggregate=1d-vec -scalarize-top-module=false` produces the following output:

```
// Generated by CIRCT 1.56.1g20231003_ce19dc1
module Foo(
  input  [1:0][1:0] in1,
                    in2,
  input             cond,
  output [1:0][1:0] o
);

  assign o = cond ? in1 : in2;
endmodule
```

We do now run `MergeConnections`, but looks like it's HW canonicalization where the array_create/two-mux/array-get simplifies down to just a single mux (!).

The reported concern appears to work now in practice, but is this something we'd like to see better handled at FIRRTL level itself?

seldridge, 2024-02-10:
Calling this closed as the reported issue is producing the desired output.

closed 2024-02-10 by seldridge