llvm/circt #3651: [HWArith] ChainedCast canonicalization pattern is too permissive

7FM, 2022-08-03:
The `ChainedCast` pattern, introduced in https://github.com/llvm/circt/pull/3647, can produce wrong and illegal results, i.e.:
```mlir
    %0 = hwarith.cast %a : (ui8) -> (si8)
    %1 = hwarith.cast %0 : (si8) -> (si10) // Sign-extension
```
is not equivalent with:
```mlir
    %0 = hwarith.cast %a : (ui8) -> (si10) // Zero-extension
```

also
```mlir
    %0 = hwarith.cast %a : (i8) -> (si8)
    %1 = hwarith.cast %0 : (si8) -> (si10) // Sign-extension
```
can not be combined to:
```mlir
    %0 = hwarith.cast %a : (i8) -> (si10)
```

mortbopet, 2022-08-03:
Looks like i was a bit too quick with that merge - thank you for catching this.
I guess a less restrictive version is chained casts of equal signedness.

7FM, 2022-08-03:
> I guess a less restrictive version is chained casts of equal signedness.

```
%0 = hwarith.cast %a: (x) -> (im)
%1 = hwarith.cast %0: (im) -> (y)
```
Yes, the immediate cast type (`im`) should be of the same signedness as `x` or signless.
I guess, this might already cover most legal cases, but the cast semantics are a bit tricky :D

teqdruid, 2023-03-28:
@mortbopet Since you reverted the breaking commit, can this issue be closed as it is no longer a bug?

closed 2023-03-28 by teqdruid