llvm/circt #3660: [PyCDE] liveOperations.count(operation.ptr) == 0 && "cannot create detached operation that already exists"

teqdruid, 2022-08-04:
Observed in the Windows Py3.9 Release with asserts build job. Non-deterministic as retrying the build didn't fail.

```
FAIL: PyCDE :: behavioral.py (16 of 24)
******************** TEST 'PyCDE :: behavioral.py' FAILED ********************
Script:
--
: 'RUN: at line 1';   C:/Users/cloudtest/AppData/Local/Temp/build-env-zia9stda/Scripts/python.exe C:/a/_work/1/s/circt/frontends/PyCDE/build/cmake_build/./bin\py-split-input-file.py C:\a\_work\1\s\circt\frontends\PyCDE\test\behavioral.py | c:\a\_work\1\s\circt\frontends\pycde\build\cmake_build\bin\filecheck.exe C:\a\_work\1\s\circt\frontends\PyCDE\test\behavioral.py
--
Exit Code: 2

Command Output (stdout):
--
$ ":" "RUN: at line 1"
$ "C:/Users/cloudtest/AppData/Local/Temp/build-env-zia9stda/Scripts/python.exe" "C:/a/_work/1/s/circt/frontends/PyCDE/build/cmake_build/./bin\py-split-input-file.py" "C:\a\_work\1\s\circt\frontends\PyCDE\test\behavioral.py"
# command stderr:
Assertion failed: liveOperations.count(operation.ptr) == 0 && "cannot create detached operation that already exists", file C:\a\_work\1\s\circt\llvm\mlir\lib\Bindings\Python\IRCore.cpp, line 973
PLEASE submit a bug report to https://github.com/llvm/llvm-project/issues/ and include the crash backtrace.
Exception Code: 0x80000003
 #0 0x00007ffa74eec305 PyInit__circt (c:\a\_work\1\s\circt\frontends\pycde\build\cmake_build\tools\circt\python_packages\circt_core\mlir\_mlir_libs\_circt.cp39-win_amd64.pyd+0x3c305)
 #1 0x00007ffa7e9b1881 (C:\Windows\System32\ucrtbase.dll+0x71881)
 #2 0x00007ffa7e9b2851 (C:\Windows\System32\ucrtbase.dll+0x72851)
 #3 0x00007ffa7e9b41b5 (C:\Windows\System32\ucrtbase.dll+0x741b5)
 #4 0x00007ffa7e9b44f1 (C:\Windows\System32\ucrtbase.dll+0x744f1)
 #5 0x00007ffa7570f084 PyInit__mlir (c:\a\_work\1\s\circt\frontends\pycde\build\cmake_build\tools\circt\python_packages\circt_core\mlir\_mlir_libs\_mlir.cp39-win_amd64.pyd+0x5f084)
 #6 0x00007ffa75710b53 PyInit__mlir (c:\a\_work\1\s\circt\frontends\pycde\build\cmake_build\tools\circt\python_packages\circt_core\mlir\_mlir_libs\_mlir.cp39-win_amd64.pyd+0x60b53)
 #7 0x00007ffa75712ace PyInit__mlir (c:\a\_work\1\s\circt\frontends\pycde\build\cmake_build\tools\circt\python_packages\circt_core\mlir\_mlir_libs\_mlir.cp39-win_amd64.pyd+0x62ace)
 #8 0x00007ffa757441d2 PyInit__mlir (c:\a\_work\1\s\circt\frontends\pycde\build\cmake_build\tools\circt\python_packages\circt_core\mlir\_mlir_libs\_mlir.cp39-win_amd64.pyd+0x941d2)
 #9 0x00007ffa75744284 PyInit__mlir (c:\a\_work\1\s\circt\frontends\pycde\build\cmake_build\tools\circt\python_packages\circt_core\mlir\_mlir_libs\_mlir.cp39-win_amd64.pyd+0x94284)
#10 0x00007ffa756bd0fa pybind11::error_already_set::operator=(class pybind11::error_already_set &&) (c:\a\_work\1\s\circt\frontends\pycde\build\cmake_build\tools\circt\python_packages\circt_core\mlir\_mlir_libs\_mlir.cp39-win_amd64.pyd+0xd0fa)
#11 0x00007ffa654f5222 (C:\cibw\python\python.3.9.7\tools\python39.dll+0x25222)
#12 0x00007ffa65511b2c (C:\cibw\python\python.3.9.7\tools\python39.dll+0x41b2c)
#13 0x00007ffa65655bf9 (C:\cibw\python\python.3.9.7\tools\python39.dll+0x185bf9)
#14 0x00007ffa654faaa3 (C:\cibw\python\python.3.9.7\tools\python39.dll+0x2aaa3)
#15 0x00007ffa654fc487 (C:\cibw\python\python.3.9.7\tools\python39.dll+0x2c487)
#16 0x00007ffa654f2041 (C:\cibw\python\python.3.9.7\tools\python39.dll+0x22041)
#17 0x00007ffa654f23ad (C:\cibw\python\python.3.9.7\tools\python39.dll+0x223ad)
#18 0x00007ffa756b7746 pybind11::error_already_set::operator=(class pybind11::error_already_set &&) (c:\a\_work\1\s\circt\frontends\pycde\build\cmake_build\tools\circt\python_packages\circt_core\mlir\_mlir_libs\_mlir.cp39-win_amd64.pyd+0x7746)
#19 0x00007ffa654e7858 (C:\cibw\python\python.3.9.7\tools\python39.dll+0x17858)
#20 0x00007ffa65503773 (C:\cibw\python\python.3.9.7\tools\python39.dll+0x33773)
#21 0x00007ffa654faaa3 (C:\cibw\python\python.3.9.7\tools\python39.dll+0x2aaa3)
#22 0x00007ffa654fc487 (C:\cibw\python\python.3.9.7\tools\python39.dll+0x2c487)
#23 0x00007ffa654f5c8f (C:\cibw\python\python.3.9.7\tools\python39.dll+0x25c8f)
#24 0x00007ffa654e7908 (C:\cibw\python\python.3.9.7\tools\python39.dll+0x17908)
#25 0x00007ffa654e77f6 (C:\cibw\python\python.3.9.7\tools\python39.dll+0x177f6)
#26 0x00007ffa65503773 (C:\cibw\python\python.3.9.7\tools\python39.dll+0x33773)
#27 0x00007ffa654faaa3 (C:\cibw\python\python.3.9.7\tools\python39.dll+0x2aaa3)
#28 0x00007ffa654fc487 (C:\cibw\python\python.3.9.7\tools\python39.dll+0x2c487)
#29 0x00007ffa654ff2aa (C:\cibw\python\python.3.9.7\tools\python39.dll+0x2f2aa)
#30 0x00007ffa654ff1cc (C:\cibw\python\python.3.9.7\tools\python39.dll+0x2f1cc)
#31 0x00007ffa654ff9ea (C:\cibw\python\python.3.9.7\tools\python39.dll+0x2f9ea)
#32 0x00007ffa654fc3a4 (C:\cibw\python\python.3.9.7\tools\python39.dll+0x2c3a4)
#33 0x00007ffa654ff2aa (C:\cibw\python\python.3.9.7\tools\python39.dll+0x2f2aa)
#34 0x00007ffa654faaa3 (C:\cibw\python\python.3.9.7\tools\python39.dll+0x2aaa3)
#35 0x00007ffa654fc487 (C:\cibw\python\python.3.9.7\tools\python39.dll+0x2c487)
#36 0x00007ffa654ff2aa (C:\cibw\python\python.3.9.7\tools\python39.dll+0x2f2aa)
#37 0x00007ffa654ff1cc (C:\cibw\python\python.3.9.7\tools\python39.dll+0x2f1cc)
#38 0x00007ffa654faaa3 (C:\cibw\python\python.3.9.7\tools\python39.dll+0x2aaa3)
#39 0x00007ffa6550045a (C:\cibw\python\python.3.9.7\tools\python39.dll+0x3045a)
#40 0x00007ffa654faaa3 (C:\cibw\python\python.3.9.7\tools\python39.dll+0x2aaa3)
#41 0x00007ffa655014a3 (C:\cibw\python\python.3.9.7\tools\python39.dll+0x314a3)
#42 0x00007ffa654faaa3 (C:\cibw\python\python.3.9.7\tools\python39.dll+0x2aaa3)
#43 0x00007ffa65514735 (C:\cibw\python\python.3.9.7\tools\python39.dll+0x44735)
#44 0x00007ffa6551606f (C:\cibw\python\python.3.9.7\tools\python39.dll+0x4606f)
#45 0x00007ffa655160b1 (C:\cibw\python\python.3.9.7\tools\python39.dll+0x460b1)
#46 0x00007ffa655184e2 (C:\cibw\python\python.3.9.7\tools\python39.dll+0x484e2)
#47 0x00007ffa65518462 (C:\cibw\python\python.3.9.7\tools\python39.dll+0x48462)
#48 0x00007ffa654dee17 (C:\cibw\python\python.3.9.7\tools\python39.dll+0xee17)
#49 0x00007ffa6551784a (C:\cibw\python\python.3.9.7\tools\python39.dll+0x4784a)
#50 0x00007ffa6551769f (C:\cibw\python\python.3.9.7\tools\python39.dll+0x4769f)
#51 0x00007ffa654fe83e (C:\cibw\python\python.3.9.7\tools\python39.dll+0x2e83e)
#52 0x00007ffa654faaa3 (C:\cibw\python\python.3.9.7\tools\python39.dll+0x2aaa3)
#53 0x00007ffa65514735 (C:\cibw\python\python.3.9.7\tools\python39.dll+0x44735)
#54 0x00007ffa6551606f (C:\cibw\python\python.3.9.7\tools\python39.dll+0x4606f)
#55 0x00007ffa655160b1 (C:\cibw\python\python.3.9.7\tools\python39.dll+0x460b1)
#56 0x00007ffa655184e2 (C:\cibw\python\python.3.9.7\tools\python39.dll+0x484e2)
#57 0x00007ffa65518462 (C:\cibw\python\python.3.9.7\tools\python39.dll+0x48462)
#58 0x00007ffa655579db (C:\cibw\python\python.3.9.7\tools\python39.dll+0x879db)
#59 0x00007ffa65557d6c (C:\cibw\python\python.3.9.7\tools\python39.dll+0x87d6c)
#60 0x00007ffa655d911b (C:\cibw\python\python.3.9.7\tools\python39.dll+0x10911b)
#61 0x00007ffa655d91c0 (C:\cibw\python\python.3.9.7\tools\python39.dll+0x1091c0)
#62 0x00007ffa6555a642 (C:\cibw\python\python.3.9.7\tools\python39.dll+0x8a642)
#63 0x00007ffa6555b524 (C:\cibw\python\python.3.9.7\tools\python39.dll+0x8b524)
#64 0x00007ffa6555b3ad (C:\cibw\python\python.3.9.7\tools\python39.dll+0x8b3ad)
#65 0x00007ffa6555add5 (C:\cibw\python\python.3.9.7\tools\python39.dll+0x8add5)
#66 0x00007ff7a4441254 (C:\cibw\python\python.3.9.7\tools\python.exe+0x1254)
#67 0x00007ffa80687034 (C:\Windows\System32\KERNEL32.DLL+0x17034)
#68 0x00007ffa810c2651 (C:\Windows\SYSTEM32\ntdll.dll+0x52651)

error: command failed with exit status: 2147483651
$ "c:\a\_work\1\s\circt\frontends\pycde\build\cmake_build\bin\filecheck.exe" "C:\a\_work\1\s\circt\frontends\PyCDE\test\behavioral.py"
# command stderr:
FileCheck error: '<stdin>' is empty.
FileCheck command line:  c:\a\_work\1\s\circt\frontends\pycde\build\cmake_build\bin\filecheck.exe C:\a\_work\1\s\circt\frontends\PyCDE\test\behavioral.py

error: command failed with exit status: 2

--

********************
Testing:  0.. 10.. 20.. 30.. 40.. 50.. 60.. 70.. 80.. 90.. 
********************
Failed Tests (1):
  PyCDE :: behavioral.py

```

This is unfortunately the exact same signature of the old Python/C++ Operation pointer sync issue. I thought we had this figured out but apparently not, @mikeurbach . Unsure if the behavioral test failing or not is a red herring. I'm doing some pretty funky stuff with stack frames, so its probably not.

mikeurbach, 2022-08-04:
*sigh* i remember this error signature... it's good that you seem to have it pinned down to the recent thing, maybe some line is being crossed there? i can try to look at that change more deeply.

teqdruid, 2022-08-04:
So I've been able to repro about 1/5 of the time but only with the same configuration: Py39 on Windows. Added some debug prints and started manually clearing the liveops cache between tests and it no longer crashes. Seems like something is holding on to op refs. This happens on all platforms (AFAICT), though it only causes a crash on Py39 Windows.

I suspect I'm not properly dealing with a refcounter when I update the stack. The reason the warning about live ops still held in the release ops routine wasn't being printed was that it wasn't being run. Anywhere! Oops.

teqdruid, 2022-08-04:
Found and fixed. Had nothing to do with stack updates!

PR #3665 (2022-08-05): [PyCDE] Fix liveOperations bug
The `Else` global object was holding on to refs after __exit__. This
shouldn't have been a global object reference to begin with.

Fixes #3660 .
commit d9c1b935 (2022-08-05): [PyCDE] Fix liveOperations bug (#3665)

The `Else` global object was holding on to refs after __exit__. This
shouldn't have been a global object reference to begin with.

Fixes #3660.
closed 2022-08-05 by teqdruid