llvm/circt #3678: [IMCP] Suboptimal results with registers

uenoku, 2022-08-08:
```mlir
firrtl.circuit "Example"  {
  firrtl.module private @Test(in %in1: !firrtl.uint<1>, in %in2: !firrtl.uint<1>, in %clock: !firrtl.clock, out %out: !firrtl.uint<1>) {
    %r = firrtl.reg  %clock  : !firrtl.uint<1>
    %0 = firrtl.or %in1, %r : (!firrtl.uint<1>, !firrtl.uint<1>) -> !firrtl.uint<1>
    firrtl.strictconnect %out, %0 : !firrtl.uint<1>
    firrtl.strictconnect %r, %in2 : !firrtl.uint<1>
  }
  firrtl.module @Example(in %clock: !firrtl.clock, out %s: !firrtl.uint<1>) {
    %c1_ui1 = firrtl.constant 1 : !firrtl.uint<1>
    %c0_ui1 = firrtl.constant 0 : !firrtl.uint<1>
    %test_in1, %test_in2, %test_clock, %test_out = firrtl.instance test  @Test(in in1: !firrtl.uint<1>, in in2: !firrtl.uint<1>, in clock: !firrtl.clock, out out: !firrtl.uint<1>)
    firrtl.strictconnect %test_clock, %clock : !firrtl.clock
    firrtl.strictconnect %test_in1, %c0_ui1 : !firrtl.uint<1>
    firrtl.strictconnect %test_in2, %c1_ui1 : !firrtl.uint<1>
    firrtl.strictconnect %s, %test_out : !firrtl.uint<1>
  }
}
```

The current result of `-firrtl-imconstprop`:
```mlir
firrtl.circuit "Example"  {
  firrtl.module private @Test(in %in1: !firrtl.uint<1>, in %in2: !firrtl.uint<1>, in %clock: !firrtl.clock, out %out: !firrtl.uint<1>) {
    %c1_ui1 = firrtl.constant 1 : !firrtl.uint<1>
    firrtl.strictconnect %out, %c1_ui1 : !firrtl.uint<1>
  }
  firrtl.module @Example(in %clock: !firrtl.clock, out %s: !firrtl.uint<1>) {
    %c1_ui1 = firrtl.constant 1 : !firrtl.uint<1>
    %c0_ui1 = firrtl.constant 0 : !firrtl.uint<1>
    %test_in1, %test_in2, %test_clock, %test_out = firrtl.instance test  @Test(in in1: !firrtl.uint<1>, in in2: !firrtl.uint<1>, in clock: !firrtl.clock, out out: !firrtl.uint<1>)
    firrtl.strictconnect %test_clock, %clock : !firrtl.clock
    firrtl.strictconnect %test_in1, %c0_ui1 : !firrtl.uint<1>
    firrtl.strictconnect %test_in2, %c1_ui1 : !firrtl.uint<1>
    firrtl.strictconnect %s, %test_out : !firrtl.uint<1>
  }
}
```

`%test_out` must be replaced by 1 but currently not. This is because `%out` is marked as overdefined in IMCP.

darthscsi, 2022-08-08:
What happens if the folder is disabled?  I don't think we depend on invalid folding for correctness now, and it shouldn't trigger.  From that sense, the invalid lattice value is not the same thing as the invalid firrtl value, I believe.

uenoku, 2022-08-08:
> What happens if the folder is disabled? I don't think we depend on invalid folding for correctness now, and it shouldn't trigger. From that sense, the invalid lattice value is not the same thing as the invalid firrtl value, I believe.

For the program above, we can get the optimal result. However, instead we will get regression around registers whose values are actually zeros, e.g:

```
firrtl.module @Zero(in %clock: !firrtl.clock, in %c: !firrtl.uint<1>, out %o: !firrtl.uint<1>) {
    %r = firrtl.reg  %clock  : !firrtl.uint<1>
    %c0_ui1 = firrtl.constant 0 : !firrtl.uint<1>
    %0 = firrtl.mux(%c, %r, %c0_ui1) : (!firrtl.uint<1>, !firrtl.uint<1>, !firrtl.uint<1>) -> !firrtl.uint<1>
    firrtl.strictconnect %r, %0 : !firrtl.uint<1>
    firrtl.strictconnect %o, %r : !firrtl.uint<1>
  }
```

dtzSiFive, 2022-08-08:
cc #3003 (as pointed out elsewhere).

darthscsi, 2022-08-09:
> However, instead we will get regression around registers whose values are actually zeros,

This seems specifically a problem with merging lattice values for mux.