llvm/circt #3687: [Handshake] Remove StandardToHandshake-specific builders, cleanup

mortbopet, 2022-08-09:
Currently, multiple builder function of the `Handshake` dialect operations are tightly coupled with how the `StandardToHandshake` maintains intermediate state during its run. I attribute this to the fact that `StandardToHandshake` is rather "old" (CIRCT-relative) and could use some love with respect to some of the lessons we've learnt over the course of the project maturing.

Some examples, with varying degrees of severity:
- Ops which places assumptions on external state, and instead should have their builder take an extra argument:
  - https://github.com/llvm/circt/blob/main/lib/Dialect/Handshake/HandshakeOps.cpp#L232-L233 (should be a builder parameter)
  - https://github.com/llvm/circt/blob/main/lib/Dialect/Handshake/HandshakeOps.cpp#L742
  - https://github.com/llvm/circt/blob/main/lib/Dialect/Handshake/HandshakeOps.cpp#L835
- Mux ops being built with invalid inputs, which only later are resolved https://github.com/llvm/circt/blob/main/lib/Dialect/Handshake/HandshakeOps.cpp#L347-L348. I suspect this could be solved by use of the `BackedgeBuilder` in `StandardToHandshake`.
- As above, but with `ControlMergeOp` https://github.com/llvm/circt/blob/main/lib/Dialect/Handshake/HandshakeOps.cpp#L444-L445
- A weird builder (why have those arguments if some of them are expected to be empty?) https://github.com/llvm/circt/blob/main/lib/Dialect/Handshake/HandshakeOps.cpp#L924-L931
- Memory ops aren't necessarily connected to LSQs, so this is a stale comment https://github.com/llvm/circt/blob/main/lib/Dialect/Handshake/HandshakeOps.cpp#L1410-L1411

RamirezLucas, 2022-11-29:
> A weird builder (why have those arguments if some of them are expected to be empty?)

In trying to address this, I may have found the reason why this builder has these weird parameters. The following TableGen canonicalization pattern instantiates a `SinkOp`. 
https://github.com/llvm/circt/blob/93112d40df2bd95058eae32566c13733f9fa7ea1/include/circt/Dialect/Handshake/HandshakeCanonicalization.td#L37-L39

In the transpiled C++ code, the `SinkOp` instantiation for this pattern looks like
```cpp
// Rewrite
auto odsLoc = rewriter.getFusedLoc({tblgen_ops[0]->getLoc(), tblgen_ops[1]->getLoc()}); (void)odsLoc;
::circt::handshake::SinkOp tblgen_SinkOp_0;
{
    ::llvm::SmallVector<::mlir::Value, 4> tblgen_values; (void)tblgen_values;
    ::llvm::SmallVector<::mlir::NamedAttribute, 4> tblgen_attrs; (void)tblgen_attrs;
    tblgen_values.push_back((*operand.begin()));
    ::llvm::SmallVector<::mlir::Type, 4> tblgen_types; (void)tblgen_types;
    tblgen_SinkOp_0 = rewriter.create<::circt::handshake::SinkOp>(odsLoc, tblgen_types, tblgen_values, tblgen_attrs);
}
rewriter.eraseOp(op0);
```

Is this a bug/limitation of TableGen (i.e., should we just rewrite this pattern in plain C++) or is there something wrong with the pattern itself?

mortbopet, 2022-11-30:
Looking at the sink op, this doesn't look like a bug.

1. A TableGen'erated rewrite pattern assumes that the default op builders are available (or alternatively, op builders with identical signatures).
2. SinkOp has `let skipDefaultBuilders = 1;` set.
3. This is set because the sinkop currently needs to have the handshake SOST (single operation w/ size and type) attributes set (https://github.com/llvm/circt/blob/main/lib/Dialect/Handshake/HandshakeOps.cpp#L1006)
4. And so that leads us to the situation where we have to manually implement the builders to ensure that those attributes are set.

Now, i was the one who introduced the SOST stuff to handshake, and will gladly admit (with the added mlir/circt experience i now have) that it's a bad design decision. The SOST attributes/functions (https://github.com/llvm/circt/blob/main/lib/Dialect/Handshake/HandshakeOps.cpp#L49-L57) do have some merit in sharing code for a few operations wrt. printing/parsing, but that's probably it. The SOST **attributes** (`dataType, control, size`) don't bring any additional information to an operation, since these values are all able to be derived from the existing operands/result types from an operation! And hence, encoding these as attributes is redundant, adds more special-case code, and introduces a possible source for bugs (i.e. if attributes aren't updated accordingly when other parts of an operation update).

So, a better (with better implying that we're relying on autogenerated TableGen logic instead of custom code) solution here could be something like:
1. replace the assumptions on the existence of the SOST attributes to instead rely on a SOST interface (reference: https://github.com/llvm/circt/blob/main/include/circt/Dialect/Handshake/HandshakeInterfaces.td). This implies that the operations which currently have SOST attributes should instead implement an interface that exposes SOST functions (`getSize, isControl, getDatatype`).
2. Clean up HandshakeOps.cpp (now mostly a matter of removing unused code) anywhere where these SOST attributes are used.
3. Re-enable default builders anywhere possible in `HandshakeOps.td`.

PR #4416 (2022-12-09): [Handshake] Cleanup Handshake operations builders
This commit overhauls Handshake operations by
- removing the majority of custom builders
- re-enabling default operation builders in most cases,
- deleting redundant operation operands,
- deleting the `sost` namespace, whose functionality is moved to an interface (`SOSTInterface`) instead,
- and refactoring the way `ControlInterface` determines whether an operation is a control operation, without relying on a redundant operation attribute. 
 
In addition, the commit adds verification logic as well as return type inference logic for a number of  [... cut]
closed 2022-12-09 by mortbopet