llvm/circt #3778: Centos Binaries for releases

lsteveol, 2022-08-25:
In some of the latest releases I've noticed macos/ubuntu precompiled binaries (which thank you!). Is it possible to add in Centos7 to this list?

teqdruid, 2022-08-25:
Is it possible to compile against manylinux to create somewhat more distro
independent binaries?

seldridge, 2022-08-25:
The Ubuntu release is built on Ubuntu, but should be portable to CentOS/RHEL. E.g., I can run the "Ubuntu" 1.14.0 release on RHEL.

lsteveol, 2022-08-25:
Interesting, I tried that release a few weeks back and was getting a GLIB error, now I don't see that. I do have the following

```
firtool: error while loading shared libraries: libzstd.so.1: cannot open shared object file: No such file or directory
```

But this is likely something missing on our end. I'll close this for now.

seldridge, 2022-08-25:
Note: I'm not at all against building more packages of this! (It's also likely reasonable to build actual `.deb` and `.rpm` packages as opposed to tarballs.) 

My "works for me" comment wasn't particularly helpful, especially when you are seeing an actual error. 😅 

We were surprised that this seemed to work on RHEL. I was specifically using RHEL 8.5 which is pretty recent.

lsteveol, 2022-08-25:
I'm on CentOS 7 (from 2019) currently. 

If it was easy to add to the release flow, I'd love it but don't want to sound too pushy. But I do realize that starts to creep in where somone wants you to build a  VxWorks version!

fangrouli, 2023-06-09:
Hi, I am working on CentOS 6, which is where the license of my EDA tool locates. I tried with the ubuntu releases, but they do give me the GLIB errors. I also tried with the full source package, but I could not find a `bin/firtool` inside though. May I know whether there is any way to make firtool work on CentOS 6? If I need extra compiling on the circt package in order to get a `firtool` executable, how to?

lsteveol, 2023-06-09:
I have the following instructions for CentOS7 (which are mostly following what's in the CIRCT README). CentOS6 is undoubtedly a different beast, and there is going to be some extras you need to download/install. Hopefully this helps.

```
git clone git@github.com:llvm/circt.git
cd circt
git submodule init
git submodule update
mkdir llvm/build
cd llvm/build

setenv CC clang
setenv CXX clang++

cmake -G Ninja ../llvm \
   -DLLVM_ENABLE_PROJECTS="mlir" \
   -DLLVM_TARGETS_TO_BUILD="host" \
   -DLLVM_ENABLE_ASSERTIONS=ON \
   -DCMAKE_BUILD_TYPE=Release \
   -DCMAKE_EXPORT_COMPILE_COMMANDS=ON

ninja -j 8

cd ../..
mkdir build
cd build
cmake -G Ninja .. \
  -DMLIR_DIR=$PWD/../llvm/build/lib/cmake/mlir \
  -DLLVM_DIR=$PWD/../llvm/build/lib/cmake/llvm \
  -DLLVM_ENABLE_ASSERTIONS=ON \
  -DCMAKE_BUILD_TYPE=Debug \
  -DCMAKE_EXPORT_COMPILE_COMMANDS=ON
  
ninja -j 8

```

fangrouli, 2023-06-09:
Will try it out later to see whether there is any more packages need to download (the worse part is I don't have root access, and need to ask for the help from lab managers). Thank you very much!

lsteveol, 2023-06-14:
You should have the option to install most of these tools locally, so should work even if you don't have root

fangrouli, 2023-06-15:
I was referring to the ninja and cmake packages, but yeah the github repositories do not require root access.