llvm/circt #3818: [StandardToHandshake] Loop task pipelining drops threads

Dinistro, 2022-09-03:
While writing additional tests for the `std-to-handshake` pass, I uncovered an issue related to loop task pipelining. 
The test defined [here](https://github.com/llvm/circt/tree/dinistro/more-task-pipeline-tests/integration_test/Dialect/Handshake/multiple_loops) breaks down as it does not emit all results. 
Independent of the input sequence, the 6th thread is always dropped. Note that his only happens for the buffer strategies `all` and `allFIFO`.

Either this is caused by a faulty test driver or there is indeed a deeper issue in the loop protection strategy. 

CC @mortbopet

mortbopet, 2022-09-04:
I think a trace and/or running with HSDBG from CIRCT-HLS would clarify what is going wrong here.

Dinistro, 2022-09-05:
Is it possible to import foreign traces into `hsdbg`?

I exported a trace from cocotb as described in https://docs.cocotb.org/en/stable/simulator_support.html and am now inspecting it in an external tool. It seems that the driver is faulty and accidentally does send the next element, even though the previous handshake didn't occur. 
[image]

I'll try to fix this in the `cocotb` helper's file.

PR #3819 (2022-09-05): [Handshake][lit] Fix helper and add additional tests
This patch fixes an issue in the `cocotb` handshake helper that was caused by reading values that were changing. 
Furthermore adds tests that triggered this issue. 

Fixes #3818
closed 2022-09-05 by Dinistro