llvm/circt #3828: [LowerToHW] Make mux result pragmas optional

fabianschuiki, 2022-09-06:
PR #3404 introduced the emission of pragmas alongside multiplexer results:
```scala
circuit MultibitMux: 
  module MultibitMux :
    input a : UInt<1>[3]
    input sel : UInt<2>
    output b : UInt<1>
    b <= a[sel]
```
```verilog
module MultibitMux(
  ...
  wire _GEN;

  wire [2:0] _GEN_0 = {{a_2}, {a_1}, {a_0}};
  assign _GEN = _GEN_0[sel] /* cadence map_to_mux */;   /* synopsys infer_mux_override */
  assign b = &sel ? a_0 : _GEN;
endmodule
```

This happens unconditionally. The problem is that for passes/tools that would like to analyze the HW lowering from FIRRTL, information is lost: the assignment is emitted as an `sv.verbatim` operation, which essentially removes the assignment from the IR and makes it hard to analyze. https://github.com/llvm/circt/blob/b77d81787ab37151bc579f2545e2369a328d7523/lib/Conversion/FIRRTLToHW/LowerToHW.cpp#L3411-L3430

It would be great to expose this mapping through a lowering option, such that users intending to further process the HW output (without the intent to go straight to SV) can disable the pragmas and retain the assignment information.

zyedidia, 2022-09-07:
#3832 does make the assignment visible via the SV dialect, which is good enough for now, but it does still pollute the HW dialect with SV ops, resulting in something like:

```
%0 = sv.wire  : !hw.inout<i5>
%1 = hw.array_get %x[%y] {sv.attributes = #sv.attributes<[#sv.attribute<"cadence map_to_mux">], emitAsComments>} : !hw.array<8xi5>
sv.assign %0, %1 {sv.attributes = #sv.attributes<[#sv.attribute<"synopsys infer_mux_override">], emitAsComments>} : i5
%2 = sv.read_inout %0 : !hw.inout<i5>
```

Ideally there could be a solution here that doesn't involve using SV ops at the HW level.

fabianschuiki, 2022-09-08:
Yeah it would be cool if we had a mux op in `comb` that actually encoded this specific use case of providing hints for the downstream tools, such that we could work around the SV dialect. Or a lowering option for LowerToHW that would skip these attributes entirely.

uenoku, 2022-09-08:
> Or a lowering option for LowerToHW that would skip these attributes entirely.

Yeah, I think it makes sense to create an option to attach pragmas if there is actual demand.

commit 55bf9a26 (2022-09-08): [LowerToHW] Use SV attributes to annotate multibit mux pragmas (#3832)

This commit is a follow up of #3404 to replace verbatim uses with SV attributes.
Fix #3828 by eliminating verbatim from multbitmux lowerings.