llvm/circt #3856: [FIRRTL] Slow test: register-randomization.fir

teqdruid, 2022-09-09:
On Ubuntu 22.04, clang 14.0.0, in debug mode, commit 23f4505a895418f355e3c49c2a444ccf3e4c8d71:

```
Slowest Tests:
--------------------------------------------------------------------------
15.16s: CIRCT :: firtool/register-randomization.fir
1.35s: CIRCT :: Dialect/LLHD/Simulator/sim_formats.mlir
1.08s: CIRCT :: Dialect/FIRRTL/SFCTests/GrandCentralInterfaces/Wire.fir
0.90s: CIRCT :: Conversion/HandshakeToFIRRTL/test_extmemory.mlir
0.85s: CIRCT :: circt-reduce/trivial.mlir
0.68s: CIRCT :: Dialect/FIRRTL/SFCTests/invalid-reg-pass.fir
0.65s: CIRCT :: firtool/firtool.fir
0.56s: CIRCT :: Dialect/FIRRTL/SFCTests/subcircuit-flow/flow.test
0.53s: CIRCT :: Conversion/HandshakeToFIRRTL/test_mux.mlir
0.50s: CIRCT :: Dialect/LLHD/Simulator/sim_wait.mlir
0.48s: CIRCT :: Dialect/FIRRTL/errors.mlir
0.36s: CIRCT :: Dialect/FIRRTL/lower-types.mlir
0.35s: CIRCT :: Conversion/ExportVerilog/sv-dialect.mlir
0.34s: CIRCT :: Dialect/LLHD/Simulator/sim_shifts.mlir
0.32s: CIRCT :: Conversion/FIRRTLToHW/lower-to-hw.mlir
0.31s: CIRCT :: Dialect/LLHD/Simulator/sim_bit_precision_drives.mlir
0.28s: CIRCT :: Conversion/ExportVerilog/hw-dialect.mlir
0.28s: CIRCT :: Dialect/LLHD/Simulator/sim_reg.mlir
0.27s: CIRCT :: Conversion/PipelineToCalyx/convert_pipeline.mlir
0.27s: CIRCT :: Dialect/LLHD/Simulator/sim_process.mlir

Tests Times:
--------------------------------------------------------------------------
[    Range    ] :: [               Percentage               ] :: [ Count ]
--------------------------------------------------------------------------
[15.0s,16.0s) :: [                                        ] :: [  1/427]
[14.0s,15.0s) :: [                                        ] :: [  0/427]
[13.0s,14.0s) :: [                                        ] :: [  0/427]
[12.0s,13.0s) :: [                                        ] :: [  0/427]
[11.0s,12.0s) :: [                                        ] :: [  0/427]
[10.0s,11.0s) :: [                                        ] :: [  0/427]
[ 9.0s,10.0s) :: [                                        ] :: [  0/427]
[ 8.0s, 9.0s) :: [                                        ] :: [  0/427]
[ 7.0s, 8.0s) :: [                                        ] :: [  0/427]
[ 6.0s, 7.0s) :: [                                        ] :: [  0/427]
[ 5.0s, 6.0s) :: [                                        ] :: [  0/427]
[ 4.0s, 5.0s) :: [                                        ] :: [  0/427]
[ 3.0s, 4.0s) :: [                                        ] :: [  0/427]
[ 2.0s, 3.0s) :: [                                        ] :: [  0/427]
[ 1.0s, 2.0s) :: [                                        ] :: [  2/427]
[ 0.0s, 1.0s) :: [*************************************** ] :: [424/427]
--------------------------------------------------------------------------

Testing Time: 15.24s
```

Anything over about 1s is too long IMO, but this is just egregious!

seldridge, 2022-09-14:
I think this is an actual performance regression (likely the same as https://github.com/llvm/circt/issues/3885) as opposed to a slow test. 70% of the time is being spent in Export Verilog and this test runs itself 4 times.

Thanks for the report.

teqdruid, 2022-09-16:
Pulling main helped a little, but it remains painfully high:

```
Slowest Tests:
--------------------------------------------------------------------------
9.11s: CIRCT :: firtool/register-randomization.fir
1.22s: CIRCT :: Dialect/LLHD/Simulator/sim_formats.mlir
1.10s: CIRCT :: Dialect/FIRRTL/SFCTests/GrandCentralInterfaces/Wire.fir
0.87s: CIRCT :: Conversion/HandshakeToFIRRTL/test_extmemory.mlir
0.68s: CIRCT :: circt-reduce/trivial.mlir
0.68s: CIRCT :: Dialect/FIRRTL/SFCTests/invalid-reg-pass.fir
0.63s: CIRCT :: firtool/firtool.fir
0.50s: CIRCT :: Dialect/FIRRTL/SFCTests/subcircuit-flow/flow.test
0.50s: CIRCT :: Conversion/HandshakeToFIRRTL/test_mux.mlir
0.46s: CIRCT :: Dialect/LLHD/Simulator/sim_wait.mlir
0.41s: CIRCT :: Dialect/FIRRTL/errors.mlir
0.35s: CIRCT :: Dialect/FIRRTL/lower-types.mlir
0.31s: CIRCT :: Conversion/ExportVerilog/sv-dialect.mlir
0.31s: CIRCT :: Conversion/ExportVerilog/hw-dialect.mlir
0.29s: CIRCT :: Dialect/LLHD/Simulator/sim_bit_precision_drives.mlir
0.28s: CIRCT :: Dialect/FIRRTL/SFCTests/dedup.fir
0.28s: CIRCT :: Dialect/LLHD/Simulator/sim_shifts.mlir
0.27s: CIRCT :: Dialect/FIRRTL/annotations.mlir
0.27s: CIRCT :: Dialect/FIRRTL/canonicalization.mlir
0.27s: CIRCT :: Dialect/LLHD/Simulator/sim_reg.mlir
```

uenoku, 2022-09-18:
Fixed by bd75af10bb15038068e9e5562c4bed14ff595aad

closed 2022-09-18 by uenoku

teqdruid, 2022-09-19:
Can confirm. Thanks!