llvm/circt #3864: [Tool] Add a `hlstool` to avoid passing a large set of flags to `circt-opt`

Dinistro, 2022-09-12:
Similar to `firtool` it makes sense to provide a `hlstool` for the `std` to verilog flow. 
Currently, our tests require a large set of flags to `circt-opt` that will only grow further:
https://github.com/llvm/circt/pull/3825/files#diff-dd38391f7380af4055d2a84bc5dde650e1400635e2eb5441643231b153942a98R12-R17

To simplify adding additional passes and splitting existing ones into pieces, such a tool will be very helpful.

teqdruid, 2022-09-12:
If it's just the long list of options to circt-opt, have you seen pass pipelines? https://mlir.llvm.org/docs/PassManagement/#pass-pipeline-registration

dtzSiFive, 2022-09-12:
Marking as closed, `hlstool` added in #3866 .

closed 2022-09-12 by dtzSiFive