llvm/circt #3879: Forbid pushes to main branch

Dinistro, 2022-09-13:
I accidentally pushed a small commit directly to the main branch...
Would it make sense to forbid this? Going through a PR that one merges without a review is still possible, but the CI will at least run beforehand.

teqdruid, 2022-09-13:
I've proposed this in the past but been shouted down. This is actually the way the llvm monorepo works also it turns out.

Dinistro, 2022-09-13:
That seems a somewhat weird policy, but if the community prefers it this way, I'm fine with that.
I'll keep this issue open for a day or two, in case somebody really wants this to change.

mortbopet, 2022-09-14:
I'd be in favor of this. I occasionally push directly to main for tiny fixes, typos, etc.. Most of the time it works out, sometimes it doesn't. Personally, I don't find opening PRs to be too cumbersome, and passing CI should obv. be mandatory - even for the tiniest of nits. For the moment, I don't think locking main is critical due to the still relatively small developer base that actively contributes to CIRCT, but assuming continued growth, this is obviously something that must be considered in the future.

stephenneuendorffer, 2022-09-14:
I think this is a place where PRs offer something that the LLVM processes don't: good precommit testing.  I think strongly encouraging people to use PRs is a good thing.  That said, I also think it can slow certain things down and get in the way at times.  I don't feel the need to force people to always go through PRs, as long as the community can be trusted to wield that power carefully.

seldridge, 2022-09-14:
As long as we continue to live in a world of blameless reverts, I don't see an issue with continuing to empower developers to push to `main` directly. I admit that this runs counter to more traditional GitHub-hosted projects (e.g., Chisel requires pre-commit review and all tests to pass on top-of-tree, not just at the branch point). I came from that more traditional world and grew to really like the freedom that LLVM and CIRCT give the developers here. Give developers with a proven track record the power to move at whatever speed they want. Mistakes do happen and we allow blameless reverts for that reason.

That said, whenever I push directly to main, I triple check tests, run clang-format on all changed files, and still have sweaty palms when I do the push. 😅

I am, however, adamantly in favor of disabling any force-push to main, though. That should never happen and we should make sure that that doesn't accidentally occur.

closed 2022-09-20 by Dinistro