llvm/circt #3881: [Handshake] Add a synchronization operation for arbitrary types

Dinistro, 2022-09-13:
While trying to generalize https://github.com/llvm/circt/pull/3855 for functions without a control input I stumbled across the following limitation:
Currently, `handshake` has the `join` operation to force synchronization for `none` types, but there exists no equivalent primitive for other types. In other words, `join` allows us to construct circuits that stall the emission of a `none` type signal until some other computation has finished. 

```mlir
handshake.func(%in: i32, %ctrl: none) {
  %res = // long latency computation with %in
  %outCtrl = join %ctrl, %res : none, i32
  return %in, %outCtrl : i32, none
}
```

Having an operation that would allow us to do the same for other types can be very helpful in supporting dataflow abstractions that hold multiple threads, e.g., task pipelining or locking. 
I tried to come up with a construction that gives us the same functionality by using existing `handshake` operation. Unfortunately, I could only construct circuits that will break once additional buffers get inserted. 

Therefore, I propose to add an operation that looks as follows:

```mlir
%res = sync %data, %ctrl : (i32, none) -> i32
```

This operation only emits `%res` once both `%data` and `%ctrl` are valid.

mortbopet, 2022-09-14:
I've been wondering about this myself so good to see that it's being raised as an issue.

> I tried to come up with a construction that gives us the same functionality by using existing handshake operation. Unfortunately, I could only construct circuits that will break once additional buffers get inserted.

Could you elaborate on your experiments here?

To me, the `sync` operation feels like it takes N arbitrary inputs, `join`s their control-components, and emits the `data` outputs with all of the output handshake signals attached to that join. In that sense, I think it would make sense to also emit a `none` typed signal for the example you show (`%res = sync %data, %ctrl : (i32, none) -> (i32, none`).

What irks me about that `sync` operator is that it's a strictly control-only operation that just wraps a join. That makes me think whether a form of "explode" operator (and a similar "combine" operator) could be introduced to split data from control, and then just use a `join` on those exploded control signals.
However, this would obviously violate the core assumption of handshake that _any_ SSA value carries handshake semantics...

So, with all that being said, i'd be in favor of a `sync` operator since I don't think there's any other way around this problem. I'd then also expect that once we (in the future) figure out this as-of-yet unknown "lower" form of Handshake, that this splitting of data and control semantics in the circuit would allow the `sync` op to lower to a control-only `join` op.

... then, as a final comment - if we introduce `sync` does this not mean that `join` in some sense will be redundant? A `join` on N `none`-typed inputs should then just map to a `sync` where all but the first output values are unused.
```mlir
%out = join %a, %b, %c: none, none, none
return ..., %out

// same as
%out:3 = sync %a, %b, %c: none, none, none
return ..., %out#0
```

Dinistro, 2022-09-14:
> Could you elaborate on your experiments here?

I tried a lot of things, but nothing worked. Due to time-constraints (military-like event), I can currently not boil this down to something reasonable, sorry. 

> What irks me about that sync operator is that it's a strictly control-only operation that just wraps a join. That makes me think whether a form of "explode" operator (and a similar "combine" [...]
However, this would obviously violate the core assumption of handshake that any SSA value carries handshake semantics...

Thinking about something like this might, as you suggest, only make sense if we want to introduce a "lower handshake" dialect at some point.

There are two reasonable approaches:
1. We keep `sync` simple and do only allow two inputs, one being data the other control. 
2. We allow arbitrary inputs, which will all be forwarded once all of them are ready. 

The later is probably simpler to use, as it will not clutter the IR that much, if we have a lot of inputs.

> if we introduce sync does this not mean that join in some sense will be redundant?

Currently, a `join` operation will always emit a `none` type, even if no input is a `none` type. in other words, it's basically a `sync` followed by an "explode".

Dinistro, 2022-09-19:
I'll close this as the HW lowering is the only thing remaining to do and this is blocked by another issue: #3921

closed 2022-09-19 by Dinistro