llvm/circt #3915: [LowerSeqToSV] Move `RANDOM calls into `for` statement

uenoku, 2022-09-16:
Input:
```scala
circuit Foo:
  module Foo:
    input clock: Clock
    input d: UInt<49>
    output q: UInt<49>

    reg _r: UInt<49>, clock
    _r <= d

    reg r: UInt<49>, clock
    r <= d

    reg s: UInt<49>, clock
    s <= d

    q <= s
```
current output:
```verilog
      automatic logic [31:0] _RANDOM_0;
      automatic logic [31:0] _RANDOM_1;
      automatic logic [31:0] _RANDOM_2;
      automatic logic [31:0] _RANDOM_3;
      automatic logic [31:0] _RANDOM_4;
      `ifdef INIT_RANDOM_PROLOG_
        `INIT_RANDOM_PROLOG_
      `endif
      `ifdef RANDOMIZE_REG_INIT
        _RANDOM_0 = `RANDOM;
        _RANDOM_1 = `RANDOM;
        _RANDOM_2 = `RANDOM;
        _RANDOM_3 = `RANDOM;
        _RANDOM_4 = `RANDOM;
```

Ideally it can be expressed as 
```verilog
      automatic logic [31:0] _RANDOM [4:0];
      `ifdef INIT_RANDOM_PROLOG_
        `INIT_RANDOM_PROLOG_
      `endif
      `ifdef RANDOMIZE_REG_INIT
        for (i=0;i<4;i++)
        _RANDOM[i] = `RANDOM;
       ...
```

dtzSiFive, 2023-07-28:
Appears to be done in #4879.

dtzSiFive, 2023-07-28:
Original example produces:

(snipped)
```
      automatic logic [31:0] _RANDOM[0:4];
      `ifdef INIT_RANDOM_PROLOG_
        `INIT_RANDOM_PROLOG_
      `endif // INIT_RANDOM_PROLOG_
      `ifdef RANDOMIZE_REG_INIT
        for (logic [2:0] i = 3'h0; i < 3'h5; i += 3'h1) begin
          _RANDOM[i] = `RANDOM;
        end
        s = {_RANDOM[3'h3][31:2], _RANDOM[3'h4][18:0]};
```

closed 2023-07-28 by dtzSiFive