llvm/circt #3966: ExportVerilog crashes when encountering an seq.firrtlreg

darthscsi, 2022-09-21:
It should give a nice error about unhandled operation.

seldridge, 2022-09-21:
This does the right thing for a trivial file, but I can get a bad error for the following un-reduced error:

```mlir
module {
  hw.module private @GCD(%clock: i1, %io_e: i1) -> (io_z: i32, io_v: i1) {
    %c0_i32 = hw.constant 0 : i32
    %c64_i32 = hw.constant 64 : i32
    %c48_i32 = hw.constant 48 : i32
    %x = seq.firreg %2 clock %clock reset sync %io_e, %c64_i32 {firrtl.random_init_start = 0 : ui64} : i32
    %y = seq.firreg %4 clock %clock reset sync %io_e, %c48_i32 {firrtl.random_init_start = 32 : ui64} : i32
    %0 = comb.icmp bin ugt %x, %y {sv.namehint = "_T"} : i32
    %1 = comb.sub %x, %y {sv.namehint = "_x_T_1"} : i32
    %2 = comb.mux bin %0, %1, %x : i32
    %3 = comb.sub %y, %x {sv.namehint = "_y_T_1"} : i32
    %4 = comb.mux bin %0, %y, %3 : i32
    %5 = comb.icmp eq %y, %c0_i32 {sv.namehint = "_io_v_T"} : i32
    hw.output %x, %5 : i32, i1
  }
  hw.module @GCDTester(%clock: i1, %reset: i1) {
    %false = hw.constant false
    %c16_i32 = hw.constant 16 : i32
    %true = hw.constant true
    %dut.io_z, %dut.io_v = hw.instance "dut" @GCD(clock: %clock: i1, io_e: %first: i1) -> (io_z: i32, io_v: i1)
    %first = seq.firreg %false clock %clock reset sync %reset, %true {firrtl.random_init_start = 0 : ui64} : i1
    %0 = comb.xor bin %first, %true {sv.namehint = "_T"} : i1
    %1 = comb.and bin %0, %dut.io_v {sv.namehint = "_T_1"} : i1
    %2 = comb.xor bin %reset, %true {sv.namehint = "_T_4"} : i1
    %3 = comb.icmp ne %dut.io_z, %c16_i32 : i32
    %4 = comb.and %1, %2, %3 : i1
    %5 = comb.xor bin %reset, %true {sv.namehint = "_T_7"} : i1
    sv.ifdef  "SYNTHESIS" {
    } else {
      sv.always posedge %clock {
        sv.if %4 {
          %ASSERT_VERBOSE_COND_ = sv.macro.ref< "ASSERT_VERBOSE_COND_"> : i1
          sv.if %ASSERT_VERBOSE_COND_ {
            sv.error "Assertion failed\0A    at GCD.scala:33 assert(dut.io.z === z.U)\0A"
          }
          %STOP_COND__0 = sv.macro.ref< "STOP_COND_"> : i1
          sv.if %STOP_COND__0 {
            sv.fatal 1
          }
        }
        %STOP_COND_ = sv.macro.ref< "STOP_COND_"> : i1
        %6 = comb.and %STOP_COND_, %1, %5 : i1
        sv.if %6 {
          sv.finish 1
        }
      }
    }
    hw.output
  }
}
```

Error is:

```
# circt-opt -export-verilog GCDTester.mlir
Assertion failed: (entry != nameTable.end() && "value expected a name but doesn't have one"), function getName, file ExportVerilog.cpp, line 710.
PLEASE submit a bug report to https://github.com/llvm/llvm-project/issues/ and include the crash backtrace.
Assertion failed: (entry != nameTable.end() && "value expected a name but doesn't have one"), function getName, file ExportVerilog.cpp, line 710.
zsh: abort      circt-opt -export-verilog GCDTester.mlir
```

uenoku, 2022-09-21:
Related: https://github.com/llvm/circt/issues/3663