llvm/circt #3968: [FIRRTL] LowerTypes can Violate New Port Uniqueness Checks

seldridge, 2022-09-22:
Consider the following circuit:

```scala
circuit Bar:
  module Bar:
    input a: UInt<1>[1]
    output a_0: UInt<1>

    a_0 <= a[0]
```

When compile with `firtool`, this will currently crash due to changes added in https://github.com/llvm/circt/pull/3891:
```
# firtool Bar.fir
Bar.fir:2:10: error: 'firrtl.module' op port names should be unique
  module Bar:
         ^
Bar.fir:2:10: note: see current operation: 
"firrtl.module"() ({
^bb0(%arg0: !firrtl.uint<1>, %arg1: !firrtl.uint<1>):
  "firrtl.strictconnect"(%arg1, %arg0) : (!firrtl.uint<1>, !firrtl.uint<1>) -> ()
}) {annotations = [], portAnnotations = [[], []], portDirections = -2 : i2, portNames = ["a_0", "a_0"], portSyms = [], portTypes = [!firrtl.uint<1>, !firrtl.uint<1>], sym_name = "Bar"} : () -> ()
```

Previously, `firtool` would produce some not-so-great names:

```verilog
module Bar(
  input  a_0,
  output a_0_0);

  assign a_0_0 = a_0;
endmodule
```

The SFC produces subjectively better names:

```verilog
module Bar(
  input   a__0,
  output  a_0
);
  assign a_0 = a__0;
endmodule
```

The SFC is approximating (though not exactly in my understanding) the definition of _prefix uniqueness_ in the FIRRTL spec. It is not necessary that we implement this, just: (1) that this doesn't crash and (2) that we produce some decent names here.

dtzSiFive, 2023-07-28:
The issue title portion was addressed by removing the checks, but maybe issue for better names?

seldridge, 2023-07-28:
Switching to a better name lowering algorithm would make sense. The current lowering is exactly correct by the spec. (More likely, the spec was written to be exact to the MFC algorithm. 🙃 ) It's also cheap to implement, unlike the better naming. I'm fine to just close this. (Will do that!)

closed 2023-07-28 by seldridge