llvm/circt #3983: [ExportVerilog,i0 support] sv.passign with i0 is not commented out

mortbopet, 2022-09-23:
(under the assumption that `i0` support for `isHWIntegerType` is merged) emitting `sv.passign` with `i0` wires results in invalid SV:

```mlir
  hw.module @Top(%arg0: i0, %clk: i1) -> (out: i0) {
    %0 = sv.reg  {hw.verilogName = "_GEN"} : !hw.inout<i0>
    sv.alwaysff(posedge %clk) {
      sv.passign %0, %arg0 : i0
    }
    %1 = sv.read_inout %0 : !hw.inout<i0>
    hw.output %1 : i0
  }
```
emits as
```sv
module Top(     
  // input  /*Zero Width*/ arg0,
     input                 clk
  // output /*Zero Width*/ out
   );

  // Zero width: reg /*Zero Width*/ _GEN;       
  always_ff @(posedge clk)      
    _GEN <= arg0;       
  // Zero width: assign out = _GEN;     
endmodule
```

uenoku, 2022-09-23:
A slightly tricky part of this issue is that we have to make sure to omit always statement as well (or add `begin/end`).

uenoku, 2022-09-23:
Currently we are missing omission mechanism for sv.assign as well. An assignment to a port is specially handled.
```mlir
hw.module @Top(%arg0: i0, %clk: i1, %a: i0, %b: i1) -> (out: i0) {
    sv.assign %0, %1 : i0
    %0 = sv.wire  {hw.verilogName = "_GEN"} : !hw.inout<i0>
    %1 = sv.read_inout %0 : !hw.inout<i0>
    hw.output %1 : i0
}
```

```
module Top(     // foo.mlir:1:1
  // input  /*Zero Width*/ arg0,
     input                 clk,
  // input  /*Zero Width*/ a,
     input                 b
  // output /*Zero Width*/ out
   );

  // Zero width: wire /*Zero Width*/ _GEN;      // foo.mlir:3:10
  assign _GEN = _GEN;   // foo.mlir:2:5, :4:10
  // Zero width: assign out = _GEN;     // foo.mlir:4:10, :5:5
endmodule
```

teqdruid, 2022-09-23:
Shouldn't both of these simply get pruned out? In the case where a module has an `i0` output, just drive it with a `hw.constant 0 : i0`.

closed 2022-09-29 by mortbopet