llvm/circt #4029: [RFC] ZVC: Multi-thread Simulation Tool

ziha0wei, 2022-09-30:
# ZVC: Multi-thread Simulation Tool

We're now pride to introduce our hardware simulation tool to CIRCT community!

Based on MLIR & LLVM compiler framework and under the support of CIRCT Dialects, ZVC eats HDLs(Chisel, Verilog, for example) and then emits the executable file running on host machine(x86 and so on). It must be said that we don't intervene in work of frontend, and any input that can be converted to FIRRTL may be well accepted by ZVC in theory.

Multi-thread partition method is introduced to take full advantage of concurrency computing of host machine to accelerate speed of execution of the hardware to be simulated.

Seems like Verilator? Believe it or not, we're determined to beat it with the help of community.

## Conversion
Source HDLs are lowered to Core Dialects(HW, Seq, Comb, SV, according to the dialect diagram shown on CIRCT homepage) level upon level and then lower to LLVM IR ultimately according to our initial concepts. Along the path, there are much work to do now to implement the conversion from HW to LLVM IR directly.

Fortunately, many contributors are there to devote their intelligence and time to support it. And we'll make our contribution to this conversion without reservation.

## Multi-thread Partition
Multi-thread can speedup the simulation. Verilator supports multi-thread, and it can also be used in Chipyard repo. We test at our local machine that it can get an up-to-5x improvement on 16 threads.

In contrast to Verilator, we'd like to integrate more with CIRCT and LLVM instead of generating C++. And ZVC will try to build with multi modules class, unlike Verilator inline all submodule into Top Module's variables and functions.

## Framework
As mentioned above, ZVC stands on the giant shoulder of MLIR & LLVM, with wide dependencies covering from frontend to backend(almost all the stages of a compiler).

Besides, we have implemented some features specialized for hardware simulation. Let's have a look at the Top Module Caller.

### Top Module Caller
It's known to us all that the main function is the entry(under most circumstances) of a program. And there is no doubt that it's the CRT that calls the entry when running a program. Some jobs are handled by CRT before and after the call, too.

In the world of hardware simulation, Top Module is the equivalent to main function, while something similar to CRT should exist to invoke it. And that's the reason of existence of Top Module Caller.

Top Module Caller parses parameters(signals) required by the user-specified Top Module and then initializes them to specific value(0 by default, but may vary according to TestBench, which is still under development). The Top Module is invoked after all the preparations are ready.

Many other functionalities can be loaded into the Caller, for instance, a timer to evaluate the execution time of the hardware simulation.

A simple example of Top Module Caller generated by ZVC automatically is shown below:
```LLVM
; ModuleID = 'main.cpp'
source_filename = "main.cpp"

%TopModuleCaller = type { i32, i32, i1 }

declare i32 @convert_comb_shift(i32 %0, i32 %1, i1 %2)

define void @Entry(ptr %0) {
entry:
  %1 = getelementptr %TopModuleCaller, ptr %0, i32 0, i32 0
  store i32 0, ptr %1, align 4
  %2 = load i32, ptr %1, align 4
  %3 = getelementptr %TopModuleCaller, ptr %0, i32 0, i32 1
  store i32 0, ptr %3, align 4
  %4 = load i32, ptr %3, align 4
  %5 = getelementptr %TopModuleCaller, ptr %0, i32 0, i32 2
  store i1 false, ptr %5, align 1
  %6 = load i1, ptr %5, align 1
  %7 = call i32 @convert_comb_shift(i32 %2, i32 %4, i1 %6)
  ret void
}

define i32 @main() {
entry:
  %0 = alloca %TopModuleCaller, align 8
  call void @Entry(ptr %0)
  ret i32 0
}
```

Here, `convert_comb_shift` is the Top Module.

In detail, Top Module Caller is implemented by LLVM IR thanks to APIs provided by LLVM C interfaces and IR Builder.

## Progress By Now
ALL are still in progress, though. We can't wait to share the great concept and implementation(merely a little, well) to the community.

|           Feature            |                             Progress                                  |
|------------------------------|-----------------------------------------------------------------------|
|Conversion from HW to LLVM IR | In Progress                                                           |
|Framework: Top Module Caller  | Basic Functionality is Ready; Extensions are still In Progress        |
|Multi-thread Partition Method | In Early Design                                                       |
|Support HDL: Chisel           | In Progress; Only FIRRTL converted from Chisel is supported currently |
|Support HDL: Verilog          | May be Provided in Subsequent Versions                                |

darthscsi, 2022-10-05:
Would you be interested in giving  a talk at the Circt ODM?

ziha0wei, 2022-10-08:
> Would you be interested in giving a talk at the Circt ODM?

Thanks for your invitation. The meeting time is not suitable for us, however, as we live in GMT+8 time zone and CIRCT ODM begins at 2:00 AM in that time zone.
Maybe we can have further discussion on LLVM Forum(discourse group) or LLVM Discord or something like these.

teqdruid, 2022-10-10:
I think we all would like to understand the difference between this work and LLHD. Also, parts of your list of features already exist in CIRCT, though perhaps in a different context and do something different from what you'd need. Overall, we'd like to avoid feature duplication. It'd be a bad thing to have two different simulation systems or passes which do the exact same thing.

> Thanks for your invitation. The meeting time is not suitable for us, however, as we live in GMT+8 time zone and CIRCT ODM begins at 2:00 AM in that time zone. Maybe we can have further discussion on [...]

Completely understand. If there's a time outside of the usual ODM time which works both for you and is still reasonable for those of us on the US west coast, I think it'd be much more efficient to discuss some of these issues synchronously. In particular, @fabianschuiki is our resident simulation expert so we should find a time he could make.

AikiQuare, 2022-10-24:
> I think we all would like to understand the difference between this work and LLHD. Also, parts of your list of features already exist in CIRCT, though perhaps in a different context and do something [...]

We discussed the time, At 17:00 PST, we have 8:00 (24h) Eastern Zone here, and at the end of the month the US switches to winter time, when your time is 16:00 p.m. and we are still 8:00 am (GMT+8), or you set the time for 17:00 p.m. and we are 9:00 am, which may also be a time to consider, depending on how you arrange. We will prepare some materials for the meeting to describe our tool and some works.

teqdruid, 2022-10-26:
@fabianschuiki this seems like a reasonable time. Since you're most critical of (IMO) to have in attendance, could you schedule this? I'm mostly available aside at this time except for Mondays, but don't schedule around me.

fabianschuiki, 2022-10-31:
Yeah that sounds great!