llvm/circt #4044: [Handshake] Canonicalize pack and unpack

Dinistro, 2022-10-03:
Currently, there are no canonicalization patterns for `pack` and `unpack`.

mortbopet, 2022-10-03:
All of this would of course be mitigated if there was just a standardized way, upstream, of handling tuple packing, unpacking, extracting, exploded (with accompanying canonicalization patterns).

One option could be to move `tuple`-type and `pack/unpack` lowering out of `HandshakeToHW` and instead as have a pre-pass that converts `tuple`s to `hw.struct`s, and pack/unpack ops to their corresponding `hw` counterparts. With this, the `hw.struct_*` canonicalizers should apply.

Dinistro, 2022-10-03:
There is a rationale that explicitly states that they will never adopt tuple operations due to non generic semantics: https://mlir.llvm.org/docs/Rationale/Rationale/#tuple-types

> One option could be to move tuple-type and pack/unpack lowering out of HandshakeToHW and instead as have a pre-pass that converts tuples to hw.structs, and pack/unpack ops to their corresponding hw counterparts. With this, the hw.struct_* canonicalizers should apply.

If that can handle the control signal accordingly, we could do so. Not sure how this would work if there is a fork in between though.

mortbopet, 2022-10-03:
> There is a rationale that explicitly states that they will never adopt tuple operations due to non generic semantics: https://mlir.llvm.org/docs/Rationale/Rationale/#tuple-types

I'm aware, and it makes sense. Personally, i'd be more in favor of a (yet another) upstream dialect containing a struct type, with optional field names, and struct create/extract operations.

> If that can handle the control signal accordingly, we could do so. Not sure how this would work if there is a fork in between though.

Should be able to! Given that any value in a `handshake.func` carries handshake semantics, inserting `hw` operations is in theory therefore also supported, assuming that whatever lowering pass that follows supports the operation. `hw` struct operations aren't yet supported in `HandshakeToHW` but this is straight forward (mostly just adding a few template instantiations to the patterns) - and i will need this shortly, anyways :). 

> Not sure how this would work if there is a fork in between though.

In general, I'd expect canonicalization patterns to work on de-materialized IR.