llvm/circt #4123: error: custom op 'calyx.program' is unknown

linuxlonelyeagle, 2022-10-19:
When i use **circt-translate** i meet a problem.The command I use is this **path/to/circt-translate --export-calyx tests/backend/mlir/simple.expect**.
There is the simple.expect file.
```
calyx.program "main" {
calyx.component @main(%go: i1 {go=1}, %clk: i1 {clk=1}, %reset: i1 {reset=1}) -> (%out: i1, %done: i1 {done=1}) {
  %r1.in, %r1.write_en, %r1.clk, %r1.reset, %r1.out, %r1.done = calyx.register @r1 : i1, i1, i1, i1, i1, i1
  %_1_1.out = hw.constant 1 : i1
  calyx.wires {
    calyx.group @Group1 {
      calyx.assign %r1.in = %_1_1.out : i1
      calyx.assign %r1.write_en = %_1_1.out : i1
      calyx.group_done %r1.done : i1
    }
  }
  calyx.control {
    calyx.seq {
      calyx.enable @Group1
    }
  }
}
}
```

mikeurbach, 2022-10-19:
I believe that is an older example of Calyx MLIR dialect, which is no longer valid. As the error mentions, `calyx.program` was removed in https://github.com/llvm/circt/pull/3580. Now, the wrapper is just the standard `builtin.module`, with Calyx dialect attributes attached. To get a sense of what the IR should look like, the current tests might be helpful, for example: https://github.com/llvm/circt/blob/6c3703ebe3cb0875036b1a66b98d344087affe98/test/Dialect/Calyx/emit.mlir#L3
I believe `calyx.entrypoint` is the only required attribute, which serves the same purpose as the string attribute on the old `calyx.program`: it names the "entrypoint" component.

linuxlonelyeagle, 2022-10-19:
> I believe that is an older example of Calyx MLIR dialect, which is no longer valid. As the error mentions, `calyx.program` was removed in #3580. Now, the wrapper is just the standard [...]

thanks

closed 2022-10-19 by linuxlonelyeagle