llvm/circt #4133: Clang warnings in fresh compile

teqdruid, 2022-10-19:
```
/home/jodemme/circt/lib/Dialect/Comb/CombFolds.cpp:117:10: warning: 'ComplementMatcher' may not intend to support class template argument deduction [-Wctad-maybe-unsupported]
  return ComplementMatcher(subExpr);
         ^
/home/jodemme/circt/lib/Dialect/Comb/CombFolds.cpp:770:27: note: in instantiation of function template specialization 'm_Complement<mlir::detail::AnyCapturedValueMatcher>' requested here
    if (matchPattern(arg, m_Complement(m_Any(&subExpr)))) {
                          ^
/home/jodemme/circt/lib/Dialect/Comb/CombFolds.cpp:105:8: note: add a deduction guide to suppress this warning
struct ComplementMatcher {
       ^
1 warning generated.
[2385/3241] Building CXX object tools/circt/unittests/Dialect/Moore/CMakeFiles/CIRCTMooreTests.dir/TypesTest.cpp.o
In file included from /home/jodemme/circt/unittests/Dialect/Moore/TypesTest.cpp:11:
/home/jodemme/circt/llvm/llvm/utils/unittest/googletest/include/gtest/gtest.h:1526:11: warning: comparison of integers of different signs: 'const unsigned int' and 'const int' [-Wsign-compare]
  if (lhs == rhs) {
      ~~~ ^  ~~~
/home/jodemme/circt/llvm/llvm/utils/unittest/googletest/include/gtest/gtest.h:1553:12: note: in instantiation of function template specialization 'testing::internal::CmpHelperEQ<unsigned int, int>' requested here
    return CmpHelperEQ(lhs_expression, rhs_expression, lhs, rhs);
           ^
/home/jodemme/circt/unittests/Dialect/Moore/TypesTest.cpp:185:3: note: in instantiation of function template specialization 'testing::internal::EqHelper::Compare<unsigned int, int, nullptr>' requested here
  ASSERT_EQ(arrayType2.getSize(), 42);
  ^
/home/jodemme/circt/llvm/llvm/utils/unittest/googletest/include/gtest/gtest.h:2056:32: note: expanded from macro 'ASSERT_EQ'
# define ASSERT_EQ(val1, val2) GTEST_ASSERT_EQ(val1, val2)
                               ^
/home/jodemme/circt/llvm/llvm/utils/unittest/googletest/include/gtest/gtest.h:2040:54: note: expanded from macro 'GTEST_ASSERT_EQ'
  ASSERT_PRED_FORMAT2(::testing::internal::EqHelper::Compare, val1, val2)
                                                     ^
1 warning generated.
[2386/3241] Building CXX object tools/circt/unittests/Dialect/HW/CMakeFiles/CIRCTHWTests.dir/HWModuleTest.cpp.o
In file included from /home/jodemme/circt/unittests/Dialect/HW/HWModuleTest.cpp:14:
/home/jodemme/circt/llvm/llvm/utils/unittest/googletest/include/gtest/gtest.h:1526:11: warning: comparison of integers of different signs: 'const unsigned long' and 'const int' [-Wsign-compare]
  if (lhs == rhs) {
      ~~~ ^  ~~~
/home/jodemme/circt/llvm/llvm/utils/unittest/googletest/include/gtest/gtest.h:1553:12: note: in instantiation of function template specialization 'testing::internal::CmpHelperEQ<unsigned long, int>' requested here
    return CmpHelperEQ(lhs_expression, rhs_expression, lhs, rhs);
           ^
/home/jodemme/circt/unittests/Dialect/HW/HWModuleTest.cpp:51:3: note: in instantiation of function template specialization 'testing::internal::EqHelper::Compare<unsigned long, int, nullptr>' requested here
  ASSERT_EQ(ports.size(), 4);
  ^
/home/jodemme/circt/llvm/llvm/utils/unittest/googletest/include/gtest/gtest.h:2056:32: note: expanded from macro 'ASSERT_EQ'
# define ASSERT_EQ(val1, val2) GTEST_ASSERT_EQ(val1, val2)
                               ^
/home/jodemme/circt/llvm/llvm/utils/unittest/googletest/include/gtest/gtest.h:2040:54: note: expanded from macro 'GTEST_ASSERT_EQ'
  ASSERT_PRED_FORMAT2(::testing::internal::EqHelper::Compare, val1, val2)
                                                     ^
/home/jodemme/circt/llvm/llvm/utils/unittest/googletest/include/gtest/gtest.h:1526:11: warning: comparison of integers of different signs: 'const unsigned int' and 'const int' [-Wsign-compare]
  if (lhs == rhs) {
      ~~~ ^  ~~~
/home/jodemme/circt/llvm/llvm/utils/unittest/googletest/include/gtest/gtest.h:1553:12: note: in instantiation of function template specialization 'testing::internal::CmpHelperEQ<unsigned int, int>' requested here
    return CmpHelperEQ(lhs_expression, rhs_expression, lhs, rhs);
           ^
/home/jodemme/circt/unittests/Dialect/HW/HWModuleTest.cpp:70:3: note: in instantiation of function template specialization 'testing::internal::EqHelper::Compare<unsigned int, int, nullptr>' requested here
  ASSERT_EQ(output->getNumOperands(), 4);
  ^
/home/jodemme/circt/llvm/llvm/utils/unittest/googletest/include/gtest/gtest.h:2056:32: note: expanded from macro 'ASSERT_EQ'
# define ASSERT_EQ(val1, val2) GTEST_ASSERT_EQ(val1, val2)
                               ^
/home/jodemme/circt/llvm/llvm/utils/unittest/googletest/include/gtest/gtest.h:2040:54: note: expanded from macro 'GTEST_ASSERT_EQ'
  ASSERT_PRED_FORMAT2(::testing::internal::EqHelper::Compare, val1, val2)
                                                     ^
2 warnings generated.
[2811/3241] Building CXX object tools/circt/lib/Dialect/FIRRTL/Import/CMakeFiles/obj.CIRCTImportFIRFile.dir/FIRParser.cpp.o
/home/jodemme/circt/lib/Dialect/FIRRTL/Import/FIRParser.cpp:1243:30: warning: 'SmallVector' may not intend to support class template argument deduction [-Wctad-maybe-unsupported]
    builder.create<AttachOp>(SmallVector{dst, src});
                             ^
/home/jodemme/circt/llvm/llvm/../mlir/include/mlir/Support/LLVM.h:72:7: note: add a deduction guide to suppress this warning
class SmallVector;
      ^
1 warning generated.
[... 63 lines cut ...]
      ^
4 warnings generated.
[2972/3241] Building CXX object tools/circt/lib/Dialect/FIRRTL/Transforms/CMakeFiles/obj.CIRCTFIRRTLTransforms.dir/GrandCentral.cpp.o
/home/jodemme/circt/lib/Dialect/FIRRTL/Transforms/GrandCentral.cpp:1918:12: warning: 'Optional' may not intend to support class template argument deduction [-Wctad-maybe-unsupported]
    return Optional(id);
           ^
/home/jodemme/circt/llvm/llvm/include/llvm/ADT/APInt.h:34:29: note: add a deduction guide to suppress this warning
template <typename T> class Optional;
                            ^
1 warning generated.
[3030/3241] Building CXX object tools/circt/lib/Dialect/LLHD/IR/CMakeFiles/obj.CIRCTLLHD.dir/LLHDOps.cpp.o
/home/jodemme/circt/lib/Dialect/LLHD/IR/LLHDOps.cpp:501:3: warning: 'StringSet' may not intend to support class template argument deduction [-Wctad-maybe-unsupported]
  llvm::StringSet sigSet;
  ^
/home/jodemme/circt/llvm/llvm/include/llvm/ADT/StringSet.h:23:7: note: add a deduction guide to suppress this warning
class StringSet : public StringMap<NoneType, AllocatorTy> {
      ^
/home/jodemme/circt/lib/Dialect/LLHD/IR/LLHDOps.cpp:502:3: warning: 'StringSet' may not intend to support class template argument deduction [-Wctad-maybe-unsupported]
  llvm::StringSet instSet;
  ^
/home/jodemme/circt/llvm/llvm/include/llvm/ADT/StringSet.h:23:7: note: add a deduction guide to suppress this warning
class StringSet : public StringMap<NoneType, AllocatorTy> {
      ^
2 warnings generated.
```

seldridge, 2022-10-19:
Thanks for filing this! I think this is a dupe of: https://github.com/llvm/circt/issues/3943

These have been showing up for a while. 😢

teqdruid, 2022-10-19:
Mostly is... there are a few other sins in there though.

closed 2022-10-20 by teqdruid