llvm/circt #4136: [Handshake] Assertion in --handshake-lower-extmem-to-hw

teqdruid, 2022-10-20:
[extmem.mlir.txt](https://github.com/llvm/circt/files/9825597/extmem.mlir.txt)

```$ circt-opt extmem.mlir --handshake-lower-extmem-to-hw
circt-opt: /home/jodemme/circt/llvm/llvm/include/llvm/ADT/STLExtras.h:1159: ReferenceT llvm::detail::indexed_accessor_range_base<mlir::OperandRange, mlir::OpOperand *, mlir::Value, mlir::Value, mlir::Value>::operator[](size_t) const [DerivedT = mlir::OperandRange, BaseT = mlir::OpOperand *, T = mlir::Value, PointerT = mlir::Value, ReferenceT = mlir::Value]: Assertion `Index < size() && "invalid index for value range"' failed.
PLEASE submit a bug report to https://github.com/llvm/llvm-project/issues/ and include the crash backtrace.
Stack dump:
0.      Program arguments: circt-opt extmem.mlir --handshake-lower-extmem-to-hw
 #0 0x00005597a9d5bbea llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) /home/jodemme/circt/llvm/llvm/lib/Support/Unix/Signals.inc:569:11
 #1 0x00005597a9d5bd9b PrintStackTraceSignalHandler(void*) /home/jodemme/circt/llvm/llvm/lib/Support/Unix/Signals.inc:636:1
 #2 0x00005597a9d5a3e6 llvm::sys::RunSignalHandlers() /home/jodemme/circt/llvm/llvm/lib/Support/Signals.cpp:104:5
 #3 0x00005597a9d5c455 SignalHandler(int) /home/jodemme/circt/llvm/llvm/lib/Support/Unix/Signals.inc:407:1
 #4 0x00007f4a65bd4520 (/lib/x86_64-linux-gnu/libc.so.6+0x42520)
 #5 0x00007f4a65c28a7c __pthread_kill_implementation ./nptl/./nptl/pthread_kill.c:44:76
 #6 0x00007f4a65c28a7c __pthread_kill_internal ./nptl/./nptl/pthread_kill.c:78:10
 #7 0x00007f4a65c28a7c pthread_kill ./nptl/./nptl/pthread_kill.c:89:10
 #8 0x00007f4a65bd4476 gsignal ./signal/../sysdeps/posix/raise.c:27:6
 #9 0x00007f4a65bba7f3 abort ./stdlib/./stdlib/abort.c:81:7
#10 0x00007f4a65bba71b _nl_load_domain ./intl/./intl/loadmsgcat.c:1177:9
#11 0x00007f4a65bcbe96 (/lib/x86_64-linux-gnu/libc.so.6+0x39e96)
#12 0x00005597aa14f7e8 llvm::detail::indexed_accessor_range_base<mlir::OperandRange, mlir::OpOperand*, mlir::Value, mlir::Value, mlir::Value>::operator[](unsigned long) const /home/jodemme/circt/llvm/llvm/include/llvm/ADT/STLExtras.h:0:5
#13 0x00005597aa7559b3 circt::handshake::ExternalMemoryOp::getStorePorts() /home/jodemme/circt/lib/Dialect/Handshake/HandshakeOps.cpp:1295:22
#14 0x00005597aa88e1d6 getMemTypeForExtmem(mlir::Value) /home/jodemme/circt/lib/Dialect/Handshake/Transforms/LowerExtmemToHW.cpp:89:25
#15 0x00005597aa88d76d (anonymous namespace)::HandshakeLowerExtmemToHWPass::lowerExtmemToHW(circt::handshake::FuncOp) /home/jodemme/circt/lib/Dialect/Handshake/Transforms/LowerExtmemToHW.cpp:370:33
#16 0x00005597aa88d202 (anonymous namespace)::HandshakeLowerExtmemToHWPass::runOnOperation() /home/jodemme/circt/lib/Dialect/Handshake/Transforms/LowerExtmemToHW.cpp:113:18
#17 0x00005597ac1d5f4a mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) /home/jodemme/circt/llvm/mlir/lib/Pass/Pass.cpp:470:21
#18 0x00005597ac1d6544 mlir::detail::OpToOpPassAdaptor::runPipeline(mlir::OpPassManager&, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) /home/jodemme/circt/llvm/mlir/lib/Pass/Pass.cpp:533:16
#19 0x00005597ac1d7e0c mlir::PassManager::runPasses(mlir::Operation*, mlir::AnalysisManager) /home/jodemme/circt/llvm/mlir/lib/Pass/Pass.cpp:836:10
#20 0x00005597ac1d7d2c mlir::PassManager::run(mlir::Operation*) /home/jodemme/circt/llvm/mlir/lib/Pass/Pass.cpp:816:60
#21 0x00005597ab5126a2 performActions(llvm::raw_ostream&, bool, bool, llvm::SourceMgr&, mlir::MLIRContext*, llvm::function_ref<mlir::LogicalResult (mlir::PassManager&)>, bool, bool) /home/jodemme/circt/llvm/mlir/lib/Tools/mlir-opt/MlirOptMain.cpp:91:17
#22 0x00005597ab512309 processBuffer(llvm::raw_ostream&, std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer> >, bool, bool, bool, bool, bool, bool, llvm::function_ref<mlir::LogicalResult (mlir::PassManager&)>, mlir::DialectRegistry&, llvm::ThreadPool*) /home/jodemme/circt/llvm/mlir/lib/Tools/mlir-opt/MlirOptMain.cpp:139:12
#23 0x00005597ab5120e8 mlir::MlirOptMain(llvm::raw_ostream&, std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer> >, llvm::function_ref<mlir::LogicalResult (mlir::PassManager&)>, mlir::DialectRegistry&, bool, bool, bool, bool, bool, bool, bool)::$_0::operator()(std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer> >, llvm::raw_ostream&) const /home/jodemme/circt/llvm/mlir/lib/Tools/mlir-opt/MlirOptMain.cpp:181:12
#24 0x00005597ab511fdd mlir::LogicalResult llvm::function_ref<mlir::LogicalResult (std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer> >, llvm::raw_ostream&)>::callback_fn<mlir::MlirOptMain(llvm::raw_ostream&, std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer> >, llvm::function_ref<mlir::LogicalResult (mlir::PassManager&)>, mlir::DialectRegistry&, bool, bool, bool, bool, bool, bool, bool)::$_0>(long, std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer> >, llvm::raw_ostream&) /home/jodemme/circt/llvm/llvm/include/llvm/ADT/STLFunctionalExtras.h:45:12
#25 0x00005597ab54e759 llvm::function_ref<mlir::LogicalResult (std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer> >, llvm::raw_ostream&)>::operator()(std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer> >, llvm::raw_ostream&) const /home/jodemme/circt/llvm/llvm/include/llvm/ADT/STLFunctionalExtras.h:68:12
#26 0x00005597ab54dd35 mlir::splitAndProcessBuffer(std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer> >, llvm::function_ref<mlir::LogicalResult (std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer> >, llvm::raw_ostream&)>, llvm::raw_ostream&, bool, bool) /home/jodemme/circt/llvm/mlir/lib/Support/ToolUtilities.cpp:28:12
#27 0x00005597ab511123 mlir::MlirOptMain(llvm::raw_ostream&, std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer> >, llvm::function_ref<mlir::LogicalResult (mlir::PassManager&)>, mlir::DialectRegistry&, bool, bool, bool, bool, bool, bool, bool) /home/jodemme/circt/llvm/mlir/lib/Tools/mlir-opt/MlirOptMain.cpp:186:10
#28 0x00005597ab5112a8 mlir::MlirOptMain(llvm::raw_ostream&, std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer> >, mlir::PassPipelineCLParser const&, mlir::DialectRegistry&, bool, bool, bool, bool, bool, bool, bool) /home/jodemme/circt/llvm/mlir/lib/Tools/mlir-opt/MlirOptMain.cpp:205:10
#29 0x00005597ab511df3 mlir::MlirOptMain(int, char**, llvm::StringRef, mlir::DialectRegistry&, bool) /home/jodemme/circt/llvm/mlir/lib/Tools/mlir-opt/MlirOptMain.cpp:298:14
#30 0x00005597a9cd99f4 main /home/jodemme/circt/tools/circt-opt/circt-opt.cpp:64:23
#31 0x00007f4a65bbbd90 __libc_start_call_main ./csu/../sysdeps/nptl/libc_start_call_main.h:58:16
#32 0x00007f4a65bbbe40 call_init ./csu/../csu/libc-start.c:128:20
#33 0x00007f4a65bbbe40 __libc_start_main ./csu/../csu/libc-start.c:379:5
#34 0x00005597a9cd9855 _start (/home/jodemme/circt_build/bin/circt-opt+0x151e855)
Aborted
```

teqdruid, 2022-10-20:
Is this the correct fix: https://github.com/llvm/circt/pull/4137/commits/ca505c49f26f9427eb42874d0a34ceee3b6ebda7 ?

mortbopet, 2022-10-20:
> Is this the correct fix: https://github.com/llvm/circt/commit/ca505c49f26f9427eb42874d0a34ceee3b6ebda7 ?

almost - fixed in 5448b68ac151192bff5c4e542eee1cf55ac74231. ldCount still needs to be used as the offset in the result list.

closed 2022-10-20 by mortbopet