llvm/circt #4152: [FIRRTL][ExportVerilog] Signed multiplier doesn't emit "$signed" expressions

tymcauley, 2022-10-21:
I'm specifying a mixed signed-unsigned multiplier like this:

```firrtl
﻿circuit Mult :
  module Mult :
    input clock : Clock
    input reset : UInt<1>
    input a : SInt<34>
    input b : UInt<32>
    output y : SInt<66>

    node _y_T = cvt(b) @[main.scala 10:10]
    node _y_T_1 = mul(a, _y_T) @[main.scala 10:10]
    node _y_T_2 = tail(_y_T_1, 1) @[main.scala 10:10]
    node _y_T_3 = asSInt(_y_T_2) @[main.scala 10:10]
    y <= _y_T_3 @[main.scala 10:5]
```

When run through the SFC, I get this output:

```verilog
module Mult(
  input         clock,
  input         reset,
  input  [33:0] a,
  input  [31:0] b,
  output [65:0] y
);
  wire [32:0] _y_T = {1'b0,$signed(b)}; // @[main.scala 10:10]
  wire [66:0] _y_T_1 = $signed(a) * $signed(_y_T); // @[main.scala 10:10]
  assign y = _y_T_1[65:0]; // @[main.scala 10:10]
endmodule
```

When run through `firtool`, I get this output:

```verilog
module Mult(
  input         clock,
                reset,
  input  [33:0] a,
                b,
  output [67:0] y);

  assign y = 68'({{34{a[33]}}, a} * {{34{b[33]}}, b});  // @[main.scala:10:10]
endmodule
```

I see that `firtool` is doing the "explicit" style of sign extension here (which I honestly like), so these two designs should be identical.

I'm trying to run a formal equivalence check on some of our Chisel design between the SFC and `firtool` outputs, but the difference in multiplier styles is causing the tools to choke. Is there any way to control how `firtool` emits these signed/unsigned conversions? I also noticed `integration_test/EmitVerilog/verilog_equiv.fir`, which seems to have examples of FIRRTL generating `$signed` expressions, but for whatever reason that doesn't happen here. I tried some simple fixes (do a signed-signed multiplier, make both inputs have the same size, etc), but the `firtool` output always looked the same.

Here's how I invoked `firtool`:

```
firtool \
    --format=fir \
    --dedup \
    --verify-each \
    --split-verilog \
    --lowering-options=disallowPackedArrays,emittedLineLength=2048,noAlwaysComb,disallowLocalVariables,explicitBitcast,verifLabels,locationInfoStyle=wrapInAtSquareBracket \
    -o firtool-output \
    Mult.fir
```

Thanks for any help!

tymcauley, 2022-10-24:
Thanks for the suggestion! Yup, I was also able to get a smaller design to go through formal successfully. I guess I'm more interested to see if this is the intended behavior, or if there's any user control over this aspect of emission.

uenoku, 2023-06-02:
According to Synopsys datapath coding guidelines, SFC's output is preferable in terms of QoR. Probably we can emit $signed in the ExportVerilog by pattern matching explicit sign extension.

uenoku, 2023-07-07:
FYI `{{8{a_0[7]}}, a_0} * {{8{a_1[7]}}, a_1};` seems to consume 40% more area than `$signed(a_0) * $signed(a_1);` (with yosys) so it's definitely worth changing the emission style.