llvm/circt #4156: [ExportVerilog] Unused Ports Generate Unused Wires

seldridge, 2022-10-21:
If a submodule's output is unused we currently generate an output wire in each instantiation site for this to drive. However, I think it is reasonable to not generate this wire and instead, omit the port when hooking up the instance.

Consider:

```
circuit Foo:
  extmodule Bar:
    output a: UInt<1>
    output b: UInt<1>

  module Foo:
    output a: UInt<1>

    inst bar of Bar
    a <= bar.a
```

This currently compiles to:

```verilog
module Foo(
  output a);

  wire _bar_b;
  Bar bar (
    .a (a),
    .b (_bar_b)
  );
endmodule
```

However, it seems fine to remove `_bar_b` and instead emit:
```verilog
module Foo(
  output a);

  Bar bar (
    .a (a),
    .b (/* unused */)
  );
endmodule
```

seldridge, 2022-10-26:
Designer feedback (h/t @ingallsj) on this is that the port should just be elided. Some tools may have confusion around putting comments in ports and this avoids the problem. A reasonable alternative would be to comment out the whole port as opposed to doing the `/* unused */`, e.g., `/* .b() // unused */`. 

The general sentiment is that it is acceptable to non-novice Chisel users to have ports removed. Given this, we can start with removing the connection entirely and evolving from there.

jackkoenig, 2023-03-17:
The dangling wires also lead to LEC failures when you have an unused output of an extmodule. Because the LEC tool does not know the directionality of ports of instantiated BlackBoxes, the dangling output _could_ be a dangling input, so it is conservatively treated as non-equivalent.

It's a relatively minor issue but IMO more important than the lint problem.

mruff-sifive, 2023-03-17:
@seldridge do you know what tool is having issues with the .foo(/* unused */) form of unused port?

mruff-sifive, 2023-03-17:
VCS is vocal about missing ports on instances and is a very common place for errors, so global filtering that warning can mask real/common issues. Less so with machine generated code, but...

closed 2023-03-22 by uenoku