llvm/circt #4161: [HWMemSimImpl] Single-entry Combinational Memories Emit Mux

seldridge, 2022-10-22:
For some FIRRTL designs which include single-entry combinational memories, we emit some code which may be sub-optimal for ASIC flows.

Consider:
```
circuit Qux:
  module Qux:
    input clock: Clock
    input addr: UInt<1>
    input r: {en: UInt<1>, flip data: UInt<32>, addr: UInt<1>}
    input w: {en: UInt<1>, data: UInt<32>, addr: UInt<1>, mask: UInt<1>}

    mem m :
      data-type => UInt<32>
      depth => 1
      reader => r
      writer => w
      read-latency => 0
      write-latency => 1
      read-under-write => undefined

    m.r.clk <= clock
    m.r.en <= r.en
    m.r.addr <= r.addr
    r.data <= m.r.data

    m.w.clk <= clock
    m.w.en <= w.en
    m.w.addr <= w.addr
    m.w.data <= w.data
    m.w.mask <= w.mask
```

This currently emits a combinational memory that is, for Cadence/Synopsys, forcing a mux to exist through the use of `infer_mux_override`/`map_to_mux`:

```verilog
module m_combMem(
  input         R0_addr,
                R0_en,
                R0_clk,
                W0_addr,
                W0_en,
                W0_clk,
  input  [31:0] W0_data,
  output [31:0] R0_data);

  reg  [31:0] Memory[0:0];
  wire [31:0] _GEN;
  /* synopsys infer_mux_override */
  assign _GEN = Memory[R0_addr] /* cadence map_to_mux */;
  always @(posedge W0_clk) begin
    if (W0_en)
      Memory[W0_addr] <= W0_data;
  end // always @(posedge)
  assign R0_data = R0_en ? _GEN : 32'bx;
endmodule
```

However, this is really a single-entry memory and we can trivially elide the mux.

This type of code is coming from Chisel's `Queue` which always emits a FIRRTL memory and never opts to use a register, even for extremely small queues.

seldridge, 2022-10-22:
Note: we can patch this in the `HWMemSimImpl`. However, this may be better to handle with extensions to the FIRRTL memop canonicalizer (though this may have implications for register randomization as canonicalization runs after that). There is also a question of if this should be being handled during `MemToRegOfVec`.

mruff-sifive, 2022-10-24:
Agree we should opt. out the single entry MUX regardless of DC's issues.