llvm/circt #4181: [ExportVerilog] Expressions substituted into verbatim expressions should not wrap

dtzSiFive, 2022-10-25:
Current wrapping behavior wraps expressions if they're longer than the configured `emittedLineLength`.

This can result in verbatim statements containing newlines and indentation which does not seem in spirit of verbatim.

Test case:

```
hw.module @VerbatimWrapping(%clock : i1, %cond : i1, %val : i8, %a : i3, %b : i3) {
      %x = comb.add %a, %b : i3
      %y = comb.xor %a, %b : i3
      %arr = hw.array_create %x, %y, %x, %y, %x, %y, %x, %y, %x, %y, %x, %y : i3
      sv.verbatim "// {{0}}" (%arr) : !hw.array<12xi3>
}
```

This verbatim attempts to prefix an expression with `//` to make it a comment.

Currently this is the output of `circt-opt -export-verilog -o /dev/null`:
```
// Generated by CIRCT 1.20.0g20221025_97824a4
module VerbatimWrapping(	// ./verbatim-wrapping.mlir:1:1
  input       clock,
              cond,
  input [7:0] val,
  input [2:0] a,
              b);

  wire [2:0] _GEN = a + b;	// ./verbatim-wrapping.mlir:2:12
  wire [2:0] _GEN_0 = a ^ b;	// ./verbatim-wrapping.mlir:3:12
  // {{_GEN}, {_GEN_0}, {_GEN}, {_GEN_0}, {_GEN}, {_GEN_0}, {_GEN}, {_GEN_0}, {_GEN}, {_GEN_0},
                {_GEN}, {_GEN_0}}	// ./verbatim-wrapping.mlir:2:12, :3:12, :4:14, :5:7
endmodule
```

Which is not valid verilog due to the unexpected wrapping occurring.

dtzSiFive, 2022-10-25:
The verbatim expressions have similar possibility but since they're expected to be expressions (and are documented to already handle parentheses as appropriate) I think this is okay if not necessarily expected.

darthscsi, 2022-10-25:
verbatim statements probably should be on their one line with no indenting.

darthscsi, 2022-10-25:
I think verbatim expressions always need parenthesis.

seldridge, 2022-10-25:
😬 Nice find. That example should be entirely left alone.

Going the other direction, if a user puts a newline in an `sv.verbatim` that should be reproduced exactly. I.e., there should be no cleverness in inspecting the `sv.verbatim` to realize that it is a comment.

dtzSiFive, 2022-10-26:
Thanks all, appreciate the thoughts/feedback!  Fixed in pretty-printing draft PR, test: https://github.com/llvm/circt/blob/b9eccb96bd8a384cff4e95e0cec8926a1dee206b/test/Conversion/ExportVerilog/no-wrap.mlir (FWIW, it's basically the contents of this issue's example).

Verbatim statements [we split at newlines and indent each piece (to the "base" indentation level statements are emitted at)](https://github.com/llvm/circt/blob/fd598b33badecab72228605faa0422a7117b8743/lib/Conversion/ExportVerilog/ExportVerilog.cpp#L3019) currently, which is the preserved behavior for now.  However now each piece (and the expressions substituted in) cannot wrap and other than newlines no inspection is done.

LMK if we want to iterate/change that behavior! :+1:

PR #4194 (2022-11-21): [ExportVerilog] Use pretty-printer for improved output formatting.
Builds on top of the pretty-printing code in #4192 .

Produces much more readable in many cases (IMHO) especially with complex expressions and statements containing non-trivial expressions.

Can see before/after differences here: https://dtzsifive.github.io/circt-pp-results/ (pages may change as things develop, output not final, etc.).  Also, [chipyard](https://gist.githubusercontent.com/dtzSiFive/edde310272aff78182b2e3235c2ba6fd/) which has things like `if` statements.

Not everything is possible, but there are a lot of decisi [... cut]
closed 2022-11-21 by dtzSiFive