llvm/circt #4182: [ExportVerilog] When "commenting-out" zero width contents, wrapping may occur and break the output

dtzSiFive, 2022-10-25:
In a number of places when emitting a construct that's zero-width instead of omitting it we do so but prefix it with `//`.  This is fragile as the emitted text following can often trigger wrapping and cause the output to be invalid.

Example (modified from `hw-dialect.mlir`), `zerowidth-wrapping.mlir`:
```mlir
hw.module @TestZero(%a: i4, %zeroBitWithAVeryLongNameWhichMightSeemUnlikelyButHappensAllTheTime: i0, %arrZero: !hw.array<3xi0>)
  -> (r0: i4, rZeroOutputWithAVeryLongName_YepThisToo_LongNamesAreTheWay_MoreText_GoGoGoGoGo: i0, arrZero_0: !hw.array<3xi0>) {

  %b = comb.add %a, %a : i4
  %c = comb.add %zeroBitWithAVeryLongNameWhichMightSeemUnlikelyButHappensAllTheTime, %zeroBitWithAVeryLongNameWhichMightSeemUnlikelyButHappensAllTheTime : i0
  hw.output %b, %c, %arrZero : i4, i0, !hw.array<3xi0>
}

hw.module @TestZeroInstance(%aa: i4, %azeroBit: i0, %aarrZero: !hw.array<3xi0>)
  -> (r0: i4, rZeroOutputWithAVeryLongNameYepThisToo: i0, arrZero_0: !hw.array<3xi0>) {

  %o1, %o2, %o3 = hw.instance "iii" @TestZero(a: %aa: i4, zeroBitWithAVeryLongNameWhichMightSeemUnlikelyButHappensAllTheTime: %azeroBit: i0, arrZero: %aarrZero: !hw.array<3xi0>) -> (r0: i4, rZeroOutputWithAVeryLongName_YepThisToo_LongNamesAreTheWay_MoreText_GoGoGoGoGo: i0, arrZero_0: !hw.array<3xi0>)
 %c = comb.add %o2, %o2 : i0

  hw.output %o1, %c, %o3 : i4, i0, !hw.array<3xi0>
}
```

Here's the current output of `circt-opt -export-verilog ./zerowidth-wrapping.mlir -o /dev/null`:
```systemverilog
// Generated by CIRCT 1.20.0g20221025_97824a4
module TestZero(	// ./zerowidth-wrapping.mlir:1:1
     input  [3:0]               a,
  // input  /*Zero Width*/      zeroBitWithAVeryLongNameWhichMightSeemUnlikelyButHappensAllTheTime,
  // input  [2:0]/*Zero Width*/ arrZero,
     output [3:0]               r0
  // output /*Zero Width*/      rZeroOutputWithAVeryLongName_YepThisToo_LongNamesAreTheWay_MoreText_GoGoGoGoGo
  // output [2:0]/*Zero Width*/ arrZero_0
   );

  assign r0 = a + a;	// ./zerowidth-wrapping.mlir:4:8, :6:3
  // Zero width: assign rZeroOutputWithAVeryLongName_YepThisToo_LongNamesAreTheWay_MoreText_GoGoGoGoGo = zeroBitWithAVeryLongNameWhichMightSeemUnlikelyButHappensAllTheTime +
                zeroBitWithAVeryLongNameWhichMightSeemUnlikelyButHappensAllTheTime;	// ./zerowidth-wrapping.mlir:5:8, :6:3
  // Zero width: assign arrZero_0 = arrZero;	// ./zerowidth-wrapping.mlir:6:3
endmodule

module TestZeroInstance(	// ./zerowidth-wrapping.mlir:9:1
     input  [3:0]               aa,
  // input  /*Zero Width*/      azeroBit,
  // input  [2:0]/*Zero Width*/ aarrZero,
     output [3:0]               r0
  // output /*Zero Width*/      rZeroOutputWithAVeryLongNameYepThisToo
  // output [2:0]/*Zero Width*/ arrZero_0
   );

  // Zero width: wire /*Zero Width*/ _iii_rZeroOutputWithAVeryLongName_YepThisToo_LongNamesAreTheWay_MoreText_GoGoGoGoGo;	// ./zerowidth-wrapping.mlir:12:19
  TestZero iii (	// ./zerowidth-wrapping.mlir:12:19
    .a                                                                              (aa),
  //.zeroBitWithAVeryLongNameWhichMightSeemUnlikelyButHappensAllTheTime             (azeroBit),
  //.arrZero                                                                        (aarrZero),
    .r0                                                                             (r0)
  //.rZeroOutputWithAVeryLongName_YepThisToo_LongNamesAreTheWay_MoreText_GoGoGoGoGo (_iii_rZeroOutputWithAVeryLongName_YepThisToo_LongNamesAreTheWay_MoreText_GoGoGoGoGo)
  //.arrZero_0                                                                      (arrZero_0)
  );
  // Zero width: assign rZeroOutputWithAVeryLongNameYepThisToo = _iii_rZeroOutputWithAVeryLongName_YepThisToo_LongNamesAreTheWay_MoreText_GoGoGoGoGo +
                _iii_rZeroOutputWithAVeryLongName_YepThisToo_LongNamesAreTheWay_MoreText_GoGoGoGoGo;	// ./zerowidth-wrapping.mlir:12:19, :13:7, :15:3
endmodule
```

With the problematic lines being:

```systemverilog
  // Zero width: assign rZeroOutputWithAVeryLongName_YepThisToo_LongNamesAreTheWay_MoreText_GoGoGoGoGo = zeroBitWithAVeryLongNameWhichMightSeemUnlikelyButHappensAllTheTime +
                zeroBitWithAVeryLongNameWhichMightSeemUnlikelyButHappensAllTheTime;	// ./zerowidth-wrapping.mlir:5:8, :6:3
```

and

```systemverilog
  // Zero width: assign rZeroOutputWithAVeryLongNameYepThisToo = _iii_rZeroOutputWithAVeryLongName_YepThisToo_LongNamesAreTheWay_MoreText_GoGoGoGoGo +
                _iii_rZeroOutputWithAVeryLongName_YepThisToo_LongNamesAreTheWay_MoreText_GoGoGoGoGo;	// ./zerowidth-wrapping.mlir:12:19, :13:7, :15:3
```

I believe the instance port expression can be made to wrap as well, and perhaps others, but haven't pushed on triggering that case.

darthscsi, 2022-10-25:
would just using `/**/` be enough?

dtzSiFive, 2022-10-25:
For these cases, yes, but not if anything else introduces them within for whatever reason -- quick look through suggests we might if emitting 'debug info' (`/* inner_sym: foo */`), maybe some other cases.  Notably we also emit `/* Zero Width */` in some commented-out lines already (see above) and they don't nest.

In the short term I was planning to force these to never wrap but that isn't great although I don't expect this to be likely to happen since I think zero-width ops generally fold away (?).

I guess we could wrap in `ifdef INVALID_VERILOG` :upside_down_face: !

mortbopet, 2022-10-26:
It seems like there might be a consensus that all zero valued logic should be pruned (https://github.com/llvm/circt/pull/4176), anyways - which would mean that this issue wouldn't occur in the first place. I'd be in favor of that as well - the goal of the zero valued logic pruning pass is exactly to remove special cases like the one highlighted here (which are error prone).

https://github.com/llvm/circt/pull/4186 adds more pruning patterns (all possible i0 cases should ideally be captured by that commit). After that lands, we should be in a position where zero-width expressions won't need to be emitted as a commented out string.

dtzSiFive, 2022-10-26:
Sounds great!  FWIW fixed this in the pretty-printing work, https://github.com/llvm/circt/blob/b9eccb96bd8a384cff4e95e0cec8926a1dee206b/test/Conversion/ExportVerilog/no-wrap.mlir has the test.  However it's not great to have these 'never-break' groups and regardless what you're proposing sounds good to me! :+1: .

PR #4194 (2022-11-21): [ExportVerilog] Use pretty-printer for improved output formatting.
Builds on top of the pretty-printing code in #4192 .

Produces much more readable in many cases (IMHO) especially with complex expressions and statements containing non-trivial expressions.

Can see before/after differences here: https://dtzsifive.github.io/circt-pp-results/ (pages may change as things develop, output not final, etc.).  Also, [chipyard](https://gist.githubusercontent.com/dtzSiFive/edde310272aff78182b2e3235c2ba6fd/) which has things like `if` statements.

Not everything is possible, but there are a lot of decisi [... cut]