llvm/circt #4183: [InferResets] Not great error message for uninferred reset port

jackkoenig, 2022-10-25:
Consider:
```
circuit Top :
  extmodule Foo :
    input reset : Reset @[FizzBuzz.scala 123:10]
    input in : UInt<8>
    output out : UInt<8>

  module Top:
    input in : UInt<8>                                                                                                                                                                                             
    output out : UInt<8>

    inst foo of Foo
    foo.in <= in
    out <= foo.out
```
Run with firtool this gives the following error:
```
reset.fir:5:13: error: 'firrtl.extmodule' op contains an abstract reset type after InferResets
  extmodule Foo :
            ^
reset.fir:5:13: note: see current operation: 
"firrtl.extmodule"() ({
}) {annotations = [], parameters = [], portAnnotations = [], portDirections = -4 : i3, portNames = ["reset", "in", "out"], portSyms = [], portTypes = [!firrtl.reset, !firrtl.uint<8>, !firrtl.uint<8>], sym_name = "Foo", sym_visibility = "private"} : () -> ()
```

Not the best error, it would be ideal if it would tell me which port and to use the source locator (if available). Same thing seems to apply to ports of regular modules too.

seldridge, 2022-10-25:
We can improve this. One thing that makes the error message worse than it could be is that we don't have source locators for ports. This would likely require a `portSourceLocators` attribute array on the module. However, I don't think Chisel provides source locators for ports currently. Is this something that could be added?

closed 2022-10-25 by seldridge

seldridge, 2022-10-25:
New output:

```
issues/4183.fir:2:13: error: 'firrtl.extmodule' op has an abstract reset type port "reset" after InferResets
  extmodule Foo :
            ^
issues/4183.fir:2:13: note: see current operation: 
"firrtl.extmodule"() ({
}) {annotations = [], parameters = [], portAnnotations = [], portDirections = -4 : i3, portNames = ["reset", "in", "out"], portSyms = [], portTypes = [!firrtl.reset, !firrtl.uint<8>, !firrtl.uint<8>], sym_name = "Foo", sym_visibility = "private"} : () -> ()
```

commit b13547ee (2022-10-25): [FIRRTL] Add port name to InferResets error

Include the port name when generating an InferResets uninferred port
error.  This is the bare minimum to help debug these errors as opposed
to having to rely on the user to guess which port failed.

Port source locators are not available from Chisel (nor do we store them
in CIRCT if they did exist!), so this is the best we can do for now.

Fixes #4183.