llvm/circt #4184: [FIRRTL] Not great error message for not fully initialized wires and submodules

jackkoenig, 2022-10-25:
Consider:
```
circuit Top :
  module Foo :
    input foo : UInt<8>
    input in : UInt<8>
    output out : UInt<8>

    out <= and(in foo)

  module Top:
    input en : UInt<1>
    input in : UInt<8>
    output out : UInt<8>

    inst foo of Foo @[Top.scala 1:1]
    wire w : UInt<8> @[Top.scala 2:2]                                                                                                               
    foo.in <= in
    out <= foo.out
    when en : @[Top.scala 10:10]
      foo.foo <= UInt(0) @[Top.scala 20:20]
      w <= UInt(0) @[Top.scala 30:30]
```
This gives the error message:
```
Top.scala:1:1: error: sink "foo.foo" not fully initialized
Top.scala:1:1: note: see current operation: %0:3 = "firrtl.instance"() {annotations = [], moduleName = @Foo, name = "foo", nameKind = #firrtl<name_kind droppable_name>, portAnnotations = [[], [], []], portDirections = -4 : i3, portNames = ["foo", "in", "out"]} : () -> (!firrtl.uint<8>, !firrtl.uint<8>, !firrtl.uint<8>)
Top.scala:2:2: error: sink "w" not fully initialized
Top.scala:2:2: note: see current operation: %1 = "firrtl.wire"() {annotations = [], name = "w", nameKind = #firrtl<name_kind droppable_name>} : () -> !firrtl.uint<8>
```

It's nice that it gives the source locator and precise port, but it would be helpful to know what module I'm in (`Top`) and more context about the lack of initialization, eg. SFC gives:
```
Exception in thread "main" firrtl.passes.PassExceptions:
firrtl.passes.CheckInitialization$RefNotInitializedException:  @[Top.scala 1:1] : [module Top]  Reference foo is not fully initialized.            
   : foo.foo <= mux(en, UInt<1>("h0"), VOID)
firrtl.passes.CheckInitialization$RefNotInitializedException:  @[Top.scala 2:2] : [module Top]  Reference w is not fully initialized.              
   : w <= mux(en, UInt<1>("h0"), VOID)
firrtl.passes.PassException: 2 errors detected!
```
SFC isn't great either but it does let me know that I did at least connect to it.

seldridge, 2022-10-25:
Not a fix, but running with `--ignore-fir-locators` can improve the message slightly as you have something to search for in the .fir. The error message doesn't include any context as it can't find `Top.scala`:

```
issues/4184.fir:14:5: error: sink "foo.foo" not fully initialized
    inst foo of Foo @[Top.scala 1:1]
    ^
issues/4184.fir:14:5: note: see current operation: %0:3 = "firrtl.instance"() {annotations = [], moduleName = @Foo, name = "foo", nameKind = #firrtl<name_kind droppable_name>, portAnnotations = [[], [], []], portDirections = -4 : i3, portNames = ["foo", "in", "out"]} : () -> (!firrtl.uint<8>, !firrtl.uint<8>, !firrtl.uint<8>)
issues/4184.fir:15:5: error: sink "w" not fully initialized
    wire w : UInt<8> @[Top.scala 2:2]
    ^
issues/4184.fir:15:5: note: see current operation: %1 = "firrtl.wire"() {annotations = [], name = "w", nameKind = #firrtl<name_kind droppable_name>} : () -> !firrtl.uint<8>
```

seldridge, 2022-10-25:
New output:

```
Top.scala:1:1: error: sink "foo.foo" not fully initialized in module "Top"
Top.scala:1:1: note: see current operation: %0:3 = "firrtl.instance"() {annotations = [], moduleName = @Foo, name = "foo", nameKind = #firrtl<name_kind droppable_name>, portAnnotations = [[], [], []], portDirections = -4 : i3, portNames = ["foo", "in", "out"]} : () -> (!firrtl.uint<8>, !firrtl.uint<8>, !firrtl.uint<8>)
Top.scala:2:2: error: sink "w" not fully initialized in module "Top"
Top.scala:2:2: note: see current operation: %1 = "firrtl.wire"() {annotations = [], name = "w", nameKind = #firrtl<name_kind droppable_name>} : () -> !firrtl.uint<8>
```

commit 7567eead (2022-10-25): [FIRRTL] Print module in uninit ExpandWhens errors

Change ExpandWhens to also include the module name when printing
non-port uninitialized errors.  This is done to improve the debugging
experience from Chisel.

Fixes #4184.
closed 2022-10-25 by seldridge