llvm/circt #4201: [Handshake] Error later in pass pipeline when canonicalizers are run early on

teqdruid, 2022-10-27:
[failure.mlir.txt](https://github.com/llvm/circt/files/9882927/failure.mlir.txt)

This works:
`circt-opt --flatten-memref --handshake-legalize-memrefs --lower-std-to-handshake --handshake-lower-extmem-to-hw --canonicalize --lower-handshake-to-hw`

If I add canonicalize after `flatten-memref`, it fails:
```
circt-opt --flatten-memref --canonicalize --handshake-legalize-memrefs --lower-std-to-handshake --handshake-lower-extmem-to-hw --canonicalize --lower-handshake-to-hw
dot-cf-nonworking.mlir:5:5: error: 'handshake.constant' op result 0 has multiple uses.
    memref.store %c0_i32, %alloc[] : memref<i32>
    ^
dot-cf-nonworking.mlir:5:5: note: see current operation: %14 = "handshake.constant"(%10#2) {value = 0 : index} : (none) -> index
```

If I remove the `lower-handshake-to-hw` pass, I don't get an error:
`circt-opt --flatten-memref --canonicalize --handshake-legalize-memrefs --lower-std-to-handshake --handshake-lower-extmem-to-hw --canonicalize`

teqdruid, 2022-10-27:
There may also be other places that `canonicalize` can be inserted which causes the breakage.

mortbopet, 2022-10-28:
This is not an error, but rather a precondition of handshake hardware lowering, that all values are used exactly ones (see https://github.com/llvm/circt/blob/main/lib/Conversion/HandshakeToHW/HandshakeToHW.cpp#L1930-L1937). It is therefore required that fork/sink materialization is run before hardware lowering. This is obviously not clear from the error message, so I see how this could be confusing - will elaborate the error message, but this is nevertheless a valid error.

mortbopet, 2022-10-28:
error message elaborated in [bb6dcfc](https://github.com/llvm/circt/commit/bb6dcfc7527bdaf4be00719987cd914f99b74ef2)

closed 2022-10-28 by mortbopet