llvm/circt #4204: [ExportVerilog] Zero width error in commented out SV

teqdruid, 2022-10-28:
```mlir
hw.module @A() {
  %c = hw.constant 0 : i0
  hw.instance "b" @B(c: %c: i0) -> ()
}
hw.module @B(%c: i0) {}
```

Correctly emits:
```verilog
module A();	// spurious_error.mlir:1:1
  B b (	// spurious_error.mlir:3:3
  //.c (<<unsupported zero width constant: hw.constant>>)	// spurious_error.mlir:2:8
  );
endmodule

module B(	// spurious_error.mlir:5:1
  // input /*Zero Width*/ c
   );

endmodule
```

However, it also produces an error:

```
$ circt-opt spurious_error.mlir -export-verilog > output.sv
spurious_error.mlir:2:8: error: 'hw.constant' op will not emit zero width constants in the general case
  %c = hw.constant 0 : i0
       ^
spurious_error.mlir:2:8: note: see current operation: %0 = "hw.constant"() {value = 0 : i0} : () -> i0
```

mikeurbach, 2022-10-28:
FWIW I would also suggest that `<<unsupported zero width constant: hw.constant>>` shouldn't be emitted. It's harmless in a comment, but not a great look. That's how export verilog prints errors when it's already in the middle of exporting; something we don't usually want to see.

teqdruid, 2022-11-03:
So I'd be happy with the following fix. It'll always omit the expression, but I don't think that's terribly important. Especially since we're pruning so the logic will likely most often be ripped out.

```
diff --git a/lib/Conversion/ExportVerilog/ExportVerilog.cpp b/lib/Conversion/ExportVerilog/ExportVerilog.cpp
index 449b46eda..6deeee552 100644
--- a/lib/Conversion/ExportVerilog/ExportVerilog.cpp
+++ b/lib/Conversion/ExportVerilog/ExportVerilog.cpp
@@ -3795,7 +3795,8 @@ LogicalResult StmtEmitter::visitStmt(InstanceOp op) {
     // lowered to wire.
     OutputOp output;
     if (!elt.isOutput()) {
-      emitExpression(portVal, ops, LowestPrecedence);
+      if (!isZeroWidth)
+        emitExpression(portVal, ops, LowestPrecedence);
     } else if (portVal.hasOneUse() &&
                (output = dyn_cast_or_null<OutputOp>(
                     portVal.getUses().begin()->getOwner()))) {
```

mortbopet, 2022-11-03:
> So I'd be happy with the following fix. It'll always omit the expression, but I don't think that's terribly important. Especially since we're pruning so the logic will likely most often be ripped out.

That looks correct to me seeing as we decided on each valid user of zero-width values (InstanceOp being one of these) must explicitly handle the special case itself.

PR #4239 (2022-11-04): [ExportVerilog] Don't emit a constant if it has zero width
Fixes #4204
closed 2022-11-04 by teqdruid